EFeru/hoverboard-firmware-hack-FOC #89: Oscilation in FOC torque mode at zero input

Rudi84, 2020-08-14:
Hi Emanuel, 
after sucessfull delta motor implmentation, I also like to use some star (default motors) for some slower karts. But for zero torque input motor is oscilating for ~5-20deg in ~0.5-2Hz. If the inertia is getting bigger it reduces (10kg kart still pushed ~5cm back and forward, kart+me is ok). FOC-speed, voltage, SIN and Block is ok too. 

I found out that for lower speed Block Mode is used (en advanced mode switch), but input for Block Mode is generated by FOC (Vq). See BLDCmotorControl_FOC_R2017b_fixdt/BLDC_controller/F06_Control_Type_Management/COM_Method/
VqFin. Problem is (I think :) ) that input for Block Mode is depending on FOC PI controller --> Iq --> exact position signal (not available). Is there a possiblity to overcome this (generating input independent from FOC\ position).

For the time beeing I reduced the I gain of Iq a lot to reduce oscilation. 

Best regards
Steffen

EmanuelFeru, 2020-09-02:
Hi Steffen,

Did you check that you have a clean `0` torque input request? Because if the input is not perfect 0, it can be that the motor oscillates due to noise in the input torque request. Usually a deadband around 0, fixes the issue.

Rudi84, 2020-09-04:
Hi Emanuel, 

yes this deadband is already in for my joystick implementation and I checked also via the interface that input cmd is zero. I was also thinking about using the current measurement via the shunt. Id controller is not active in this case and then maybe assumption that complete current is Iq works ok. But shunt current is battery current not motor current. Is it possible to use somehow only the magnitude of the current vector independent from position?

Best regards
Steffen

EmanuelFeru, 2020-10-21:
>>  But shunt current is battery current not motor current. Is it possible to use somehow only the magnitude of the current vector independent from position?

Can you be more specific here to understand what you mean?

Rudi84, 2020-10-21:
Hi Emanuel,

I was only thinking loud for solutions. In the meantime I also discussed this with somebody else. And I am not sure, if this can be solved nice. And the problem is also only cosmetic. So if it is too much effort, simply ignore and close.

Problem is that in FOC-torque mode id, iq is calcclated based on the position (sin and cos of angle). For slow speed the control mode is switched but not the current calculation. So at zero/low speed current calculation for control is bad. Idea is, to maybe switch to a different position independent current calculation, where e.g. id is expected as 0 and only iq is estimated as magnitude of the current vector.

Rudi84, 2020-10-21:
Hi, I also checked the updates in your code, and I think the standstillHold function will also cover this. So can be closed.

closed 2020-10-21 by Rudi84