EFeru/hoverboard-firmware-hack-FOC #90: Torque command not respected

JulienKx, 2020-08-16:
Hi Emmanuel,

I'm back after a few month ! Thanks again for this great work !
I have upgraded my vehicule to your latest firmware and did some testing.

Here are my parameters :
FOC_CTRL
TRQ_MODE
I_MOT_MAX  5  (Low for reproducing the problem with more security while driving)    
I_DC_MAX    7 
VARIANT_ADC

I start from standstill and set the input command to maximum.
What I expect : Torque is maximum from the start, and current is 5A per motor from the start.
What is happening : Torque is low at start and increase with rpm, same for the current.

I have logged a test run :
[image]

My online Ampereter show me the same shape of graphics with current increasing with speed of the vehicule instead of being at 5A*2 immediatly.

Best Regards
Julien

EmanuelFeru, 2020-09-02:
Hi Julien,

What is time in the X axis? I want to know how slow/fast is the torque build up.
Also, in the firmware there are 2 parameters FILTER and RATE that can slow down the torque built up. These parameters are there to filter out the input and get a smoother behavior. This smoother behavior is desired since it gives better driving behavior.

Rudi84, 2020-09-04:
Hi Julien,

I think the behaviour you see is ok. Torque is proportional to phase current and there the max is constant. Because the motor inductance, caps on HW and Fet switches the system acts a bit like a DCDC converter --> the DC (battery) current is proportional to power (because battery voltage is constant). When you have constant torque and linear rising speed, your power will also rise linear. This is what you see in your current. For an ideal system current would be zero at zero speed, but we have ohmic resistance, friction, ... so there is an offset in the curve. When you put speed of wheel on x-axis you can also determine motor constants (offset - power losses, slope - speed constant).

Best regards
Steffen

JulienKx, 2020-09-07:
Hi Emanuel, Hi Steffen,

Thank you for your responses ! Here are some informations :

> What is time in the X axis? I want to know how slow/fast is the torque build up.
> Also, in the firmware there are 2 parameters FILTER and RATE that can slow down the torque built up. These parameters [...]

About the X axis : this graph represents 15 seconds.
About the parameters FILTER and RATE : on the first graph, you can see SpeedR. SpeedR value is after applying filtering and rate I think. Moreover, within 15 seconds I don't think it will behave that slow. I let them to the default values.

Steffen, your explanations about comparing with a DCDC converter make sense, but is it the desired behaviour ?
When I use another variant, for example VLT_MODE, by design the battery current is instantaneously reaching I_DC_MAX when I set the input to max. It is what I expect : the vehicule is accelerating at the maximum torque permitted by I_DC_MAX.
With TRQ_MODE, what I expect is to be able to get the same acceleration, and so to get to I_DC_MAX even at low rpms.

Best regards
Julien

Rudi84, 2020-09-22:
Hi Julien, 

if desired or not, its simply physics. Powerinput = Poweroutut. For the VLT you can be rigth, if the motor is lowohmic enough (what it is for the hoverboard). In prinzipple you give much more motorcurrent , at low speed.  In VLT_Mode, motor current is not controlled. It is only chopped, when DC_MAX current is reached. But this is no real control, it is more a protection. You should also hear this. I also had this before adjusting the parameters to my need. In chopping motor is getting realy loud and vibrating, depending how much current you chopp, you can also feel it (at least on my Bobbycar).

EmanuelFeru, 2020-10-21:
@JulienKx can you give another try with the latest commit?

JulienKx, 2020-10-30:
@EmanuelFeru  Yes I will try the latest commit. What are the changes exactly ? I see "PI controller integrator is now working correctly.".

Meanwhile, I implemented a PI controller in the main loop to control the current from the battery point of view while using torque mode. So the current in the motor can for example rise at start to 40 per motor while current from battery limited to 20 amperes per motor. Here are the settings I am testing :
// Limitation settings
#define I_MOT_MAX       40              // [A] Maximum single motor current limit
#define I_DC_MAX        22              // [A] Maximum stage2 DC Link current limit for Commutation and Sinusoidal types (This is the final current protection. Above this value, current chopping is applied. To avoid this make sure that I_DC_MAX = I_MOT_MAX + 2A)

// Current PID regulation
#define CURRENT_PID_DC_MAX  20          // [A] Maximum single motor current limit, in shunt from battery 
#define CURRENT_KP      100             // pid P params for Current
#define CURRENT_KI_X100 10*100         // pid I params for Current

closed 2021-07-24 by EmanuelFeru