EFeru/hoverboard-firmware-hack-FOC #91: Need help with wiring up variant adc with esp32

Justiudios, 2020-08-19:
Hey,
i need some help wiring up variant adc with esp32.

I connected a joystick to the left sideboard cable and can control the board. But if i replace the joystick with my esp32, it doesn't seem to react to different V inputs.
I checked the output of my GPIO25 using a multimeter and it changes the way i programmed it. (from 0 to 3.3V)
[image]

Can someone improve my wiring or help me? :)

Candas1, 2020-08-20:
Hi,

Why would you use adc between esp32 and the board, you can use USART.
There is a sample arduino sketch.

Justiudios, 2020-08-20:
Hey, 
unfortunate i didnt get it to work with uart too. 

I will give it a try again. Can you give me some information about wiring to use uart? 

Thanks

Candas1, 2020-08-20:
yes it will be a better solution.
If it's beeping it means UART communication is not ok (wrong speed, wrong frame, ... )

Justiudios, 2020-08-20:
Wow, it just works. Dont know what i did wrong last time. Thanks for the push into the right direction ^^

closed 2020-08-20 by Justiudios