EFeru/hoverboard-firmware-hack-FOC #94: VARIANT_HOVERCAR not turning on just beeping

emilhse, 2020-08-29:
Hi everyone, I'm trying to make hovercar got fresh copy made bin file uploaded into the board , it does not work it's just keep beeping. tested all connections motors connected potentiometers connected nothing help just keeps beeping and nothing works
Please advise.

Regards and thanks in advance

EmanuelFeru, 2020-08-29:
Did you calibrate the potentiometers?
Are you using a sideboard?

emilhse, 2020-08-29:
no I'm not using side boards.
I did not calibrate potentiometers.
I'm using 12 bit magnetic encoder .

do i have to connect side boards?

EmanuelFeru, 2020-08-29:
By default Hovercar expects one sideboard with the hacked firmware. If you don't want it, you need to comment out the sideboard define in config.h

And yes, you need to calibrate the limits on the ADC input otherwise it won't work. There is an auto-calibration mode, search in util.c for the function, I also gave there the procedure.

emilhse, 2020-08-29:
thank you very much

emilhse, 2020-09-11:
hi I have done everything you advised, i have commented the side boards. calibrated potentiometers. it started working but the throttle does not go to full position and still beeps unless the wheels are spinning please advise

EmanuelFeru, 2020-09-12:
Hi Emil,

In which situation does it beep? And is there a way to connect an fdti to see the values of the ADC an commands?

emilhse, 2020-09-13:
i somehow made it work by changing values  of adc in cofig file. i think there is problem auto potentiometer calibration .

emilhse, 2020-09-13:
I have ftdi adapter i will try tomorrow

emilhse, 2020-09-13:
where can i find connection diagram to connect ftdi ?

EmanuelFeru, 2020-09-13:
I don't have ftdi wiring diagram. But is very straight forward, 3 wires, gnd to and, Tx to Rx of the main board, Rx to Tx of the main board.
Very important, make sure the dip switch on the fdti, for the logic level, is at 3.3v not 5v! That's it.

When i have time I will check again the auto-calibration routine.

emilhse, 2020-09-13:
thank you very much

emilhse, 2020-09-13:
sorry for bothering you but i can not find rx tx pins on the board please help

juodumas, 2020-09-13:
Check the pinout in the README:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/raw/master/docs/pictures/mainboard_pinout.png

The Right Sideboard cable has TX/RX. Make sure not to use the 15V (usually red) wire.

emilhse, 2020-09-13:
i tried to connect using putty no luck

emilhse, 2020-09-13:
when brake potentiometer riches half of the way wheels starting to spin backwards and forward randomly

madej1, 2020-09-14:
Hi
When I've started building hoverkart, I had similar issue. When you compile the project without any changes, you will get constant beeping. Even if you will change firmware in right side board.
what I've done is to comment in config.h
  //#define SIDEBOARD_SERIAL_USART3
 // #define FEEDBACK_SERIAL_USART3
and uncomment #define DEBUG_SERIAL_USART3 
so I've got possibility to check and setup proper adc value on putty (in my case COM7 via ft232 3v3 interface)
 #define ADC1_MIN            0     
  #define ADC1_MAX            4020
and also do the battery calibration in my case
#define BAT_CALIB_REAL_VOLTAGE  3820 
#define BAT_CALIB_ADC           1441 
but also I have changed 
#define BAT_DEAD                (290 * BAT_CELLS * BAT_CALIB_ADC) / BAT_CALIB_REAL_VOLTAGE 
to 2.9V because at this level battery bms cut out the voltage to protect the battery pack.So in my case I allow BMS to take care of my battery pack instead of program algorithm. Now ...the 3-year battery lasts 15km. Obviously flux weakening off.

Still I don't figure out how to proper setup config.h in mainboard and side board firmware to show status on star 6leds  board..
BTW great project. In Poland secondhand hoverboard 30$ , sony play station steering wheel with two pedals 12$. Old kettler ketcar 27$
Best Regards
Robert,

EmanuelFeru, 2020-09-15:
Hi, I also wrote this in the Telegram group.

• Both pedals need to be connected for Hovercar, because it detects if any pedal is disconnected leading to beeping.

• Good practice to do in general is to test with ADC variant first

• If Variant ADC is working, then also VAriant Hovercar should work.

• Few things to look for (Variant_Hovercar) has some extras:
1. If no sideboard is connected -> comment-out these , otherwise it will beep.
2. Some pedals are really close to the margins, leaving no room for ADC protection -> Also lead to beeping. Try disabling the ADC protection here first. It is good to have it though in case wire gets cut or shorted for some reason. So, consider adjusting the HW, so there is space for the protection. By space I mean ADC range in the upper and lower side. This could require adjusting the potentiometer or moving slightly the hall sensor inside the pedal.

• ADC calibration is a must for Hovercar to have correct behavior and no beeping.

• The reverse in Hovercar is by double tapping the brake pedal

emilhse, 2020-09-15:
> Hi
> When I've started building hoverkart, I had similar issue. When you compile the project without any changes, you will get constant beeping. Even if you will change firmware in right side board. [...]

thanks for reply.Robert which software do you use for ft232?

i tried putty but it did not communicate. 
thanks in advance 
Emil

emilhse, 2020-09-15:
> Hi, I also wrote this in the Telegram group.
> 
> • Both pedals need to be connected for Hovercar, because it detects if any pedal is disconnected leading to beeping.
> 
> • Good practice to do in general is to test with ADC variant first
> 
> • If Variant ADC is working, then also VAriant Hovercar should work.
> 
> • Few things to look for (Variant_Hovercar) has some extras:
> 
> 1. If no sideboard is connected -> [comment-out these](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/2890745fbb5879b52821231b53c217f886344f4b/Inc/config.h#L385-L386) , otherwise it will beep.
> 2. If sideboard is connected, the sideboard needs to be flashed with the hacked version VARIANT_HOVERBOARD -> otherwise beeping. The star battery indicator (6 leds) will work automatically, no need to do anything in config.h Just flash the correct Variant.
> 3. Some pedals are really close to the margins, leaving no room for ADC protection -> Also lead to beeping. Try [disabling the ADC protection here first](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/2890745fbb5879b52821231b53c217f886344f4b/Inc/config.h#L374). It is good to have it though in case wire gets cut or shorted for some reason. So, consider adjusting the HW, so there is space for the protection. By space I mean ADC range in the upper and lower side. This could require adjusting the potentiometer or moving slightly the hall sensor inside the pedal.
> 
> • ADC calibration is a must for Hovercar to have correct behavior and no beeping.
> 
> • The reverse in Hovercar is by double tapping the brake pedal
> 
> • If you get movement or some sort of oscillations with **wheels off the ground** when the Brake pedal is pressed, **that is ok**. It is due to the fact that HOVERCAR uses Torque mode and the inertia of the system is very small when wheels are not touching the ground. Once the wheels are **on the ground** the motors will operate properly with no oscillations.

Thanks Emanuel. I will try all above.and report back

emilhse, 2020-09-17:
i made it work but only after i switched to voltage mode. it works good. i have another question how to make brake a little more stronger?

EmanuelFeru, 2020-09-18:
The motor brake is not strong enough? In my hovercar the motor brakes are quite powerful, it stops me in no time, and I weigh 77 kg.

emilhse, 2020-09-20:
one more last question, im using 48v battery the max rpm is 475-478 is there any way to increase it?

EmanuelFeru, 2020-09-20:
Yes, by activating field weakening.

emilhse, 2020-09-20:
i did already it does not go up

EmanuelFeru, 2020-09-21:
LCD I have never connected. The functionality is there from the original firmware and normally should work but I cannot guide you through that.

Regarding the Field weakening if active, also the input should now  increase up to 1500 instead of 1000. I am thinking to change the default behaviour to fully blended in the future commits.

emilhse, 2020-09-22:
Emanuel, thanks for response. 
Is there any way to connect reverse function on separate input (let's say to right sensor cable)?

Thanks in advance!!!

EmanuelFeru, 2020-09-22:
Yes, you can do that with a sideboard. I already do something similar in the Hovercar variant. I use the 2 photo switches as buttons, one for changing control mode and one to activate/deactivate field weakening. 
Communication is UART.

emilhse, 2020-09-22:
i think it should be easy. I'm weak at programming stuff. but this should be easy. right now i want to try following.
i want to connect potentiometers to Arduino make analog to serial conversion and have as much buttons as i want. i will split(by program) low and high side of pots. and then send the data via serial communication to board I"m planning also to make pedal assisted bike. I will get 2 readings , 1 from the crank the second from the rear wheel to compare speed and using arduino control the level off throttle. ( i want to put 3 levels of assist in place) so far im struggling to make serial communication to work for me.
can you please advise what do you think of this idea?

Regards
Emil

EmanuelFeru, 2020-09-23:
What issue do you have with serial? Is the example sketch working?

emilhse, 2020-09-23:
I'm going to test it today. I don't know what type of data should i send. should it be integers same as the readings of pots?
or as PWM from 0 to 255?

I can also leave everything the same( in Hovercar mode)  and simply use analogWrite function from arduino, using RC filter just set up thresh hold for reverse. 
Regards
Emil

emilhse, 2020-10-10:
can anybody help me 
?
I need to know what type of command i have to send via serial communication

thanks in advance
Regards
Emil

juodumas, 2020-10-10:
You send binary encoded shorts. Here is an example Python script that I have used when testing `VARIANT_USART`. It always sends cmd=50 steer=0:

```python
import serial
import struct
import time

key = 0xABCD
baudrate = 38400
port = '/dev/ttyUSB0'
send_interval = 100  # ms
in_data = b''

s = serial.Serial(port, baudrate)

last_sent_at = 0

while True:
    now = time.time() * 1000
    if last_sent_at == 0:
        last_sent_at = now

    in_data += s.read()

    if now - last_sent_at >= send_interval:
        last_sent_at = now
        print(in_data)
        in_data = b''
        steer = 0
        speed = 50
        checksum = key ^ steer ^ speed
        out_data = struct.pack('HhhH', key, steer, speed, checksum)
        print(s.write(out_data), out_data)
```

emilhse, 2020-10-11:
Thank you very much for your reply. do you know if there is a way to have reverse and brake commands to send via serial?

Regards
Emil

juodumas, 2020-10-12:
`cmd` & `steer` that you send via serial are signed. So just send with a minus sign for reverse or forced breaking. Or send `cmd=0` & `steer=0` for regenerative breaking.

emilhse, 2020-10-12:
> `cmd` & `steer` that you send via serial are signed. So just send with a minus sign for reverse or forced breaking. Or send `cmd=0` & `steer=0` for regenerative breaking.

thank you very much

closed 2020-10-21 by EmanuelFeru