EFeru/hoverboard-firmware-hack-FOC #96: Support for Dual board GD32F130 based Hoverbards?

ikout, 2020-09-02:
Hello,

Does this firmware work with dual based GD32F130xx boards?
Like these at this one: https://github.com/flo199213/Hoverboard-Firmware-Hack-Gen2

Thanks!

EmanuelFeru, 2020-09-03:
Not at the moment. Maybe someone can port the code to the dual board. Unfortunately, i don't have them.

happytm, 2020-09-03:
@ikout  A while back there was discussion about this here: https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/issues/82. Most new inexpensive hoverboards seems to have this kind of boards.

Thanks.

ikout, 2020-09-03:
@happytm Yeah, I have a similar one that has the same boards as also mentioned above.
The only problem that I have is that the above firmware is build with Keil and I have a Mac so I was looking for alternative firmwares for these boards that use platformIO and this lead me here.

Since I only want to control the motors using the existing hardware as you mentioned at the #82 and Arduino-FOC that would be an option or if this firmware was ported to these boards... :) 

I guess the search is on....

EmanuelFeru, 2020-10-21:
Unfortunately, dual board is not supported.

closed 2020-10-21 by EmanuelFeru