EFeru/hoverboard-firmware-hack-FOC #97: Very little torque at low RPM, stalling motors.

sugeknowles, 2020-09-09:
I am running the firmware as follows:

FOC Control, Voltage Mode
VARIANT_USART

When I try and run the motors at low (ish) RPM there is virtually no torque and the wheels will stall when under no-load.  Any speed request below 90 and the wheels simply don't move.  Being BLDC motors, they should be capable of considerable torque even at very low RPM.  Any guidance is greatly appreciated.

This is the drive command that I am sending serially (Speed = 90, Steer = 0) to the hoverboard.  
HEX:  ['0xcd', '0xab', '0x0', '0x0', '0x5a', '0x0', '0x97', '0xab']
Decimal Equivalent:  [205, 171, 0, 0, 90, 0, 151, 171]

Any thoughts?  Is this an issue with my hoverboard?  The motors?  Do I need to tune something?

EmanuelFeru, 2020-09-10:
The behavior you see is normal. In Voltage mode, requesting 90, in a range of up 1000 means 9%, so really low meaning low torque.

My suggestion is: if you are interested in torque use Torque mode, if you are interested in keeping a constant speed no matter the torque then use Speed mode.

EmanuelFeru, 2020-09-10:
No problem. We all had a learning curve :relaxed:
And you are right a proper voltage control would mean to regulate the duty cycle. But in this case Voltage mode just means a constant duty cycle to the motors.

closed 2020-09-18 by EmanuelFeru