EFeru/hoverboard-firmware-hack-FOC #98: Very fast battery drainage

kulesiax, 2020-09-13:
Hello. First of all I'd like to thank you for your great work, it's awesome what kind of things people can do with those hoverboard parts.
So let's go straight to the subject. I made an electric scooter for my niece. I did research and found out the cheapest way to make it drive would be to buy a hoverboard and use its motors and modify mainboard software. Everything went fine, I managed to make it work and drive, I tweaked the parameters like potentiometer range in your firmware to match my setup and it's working quite well. The motors run smoothly and quietly, top speed is 16km/h which is more than enough for 7 year old so as you can see I didn't put too much stress on the motors and board itself. I'm running the software in torque mode, with no field weakening etc. Most of the parameters I used are default (things like filters, current limit, max RPM and so on). The problem is that the battery runs out really, really fast. I get it that original hoverboard batteries are not too big (10s2p - 36V, 4Ah) but I read that in hoverboard they can make the board go up to 15 kilometres on one charge. For me it is 1,5km-ish max. Literally I did a test run with full battery for 15 minutes and it started beeping. In the beginning it beeped only with full throttle on, then more frequently and after few seconds when I stopped, the board turned itself off. The guy who selled the hoverboard to me claimed it was used few times only and he took care of the battery by charging it once in a while. The board looked really well. It had one tiny little scratch at the bottom so I assume he didn't lie to me. The scooter got heavier twice when I mounted the motors so the construction is light, I myself weight 70km. The motors get like a LITTLE warm after a ride, I can barely feel it but I think it's normal for the motor to get warmer when it's working. Everything seems fine except for the battery life. So the question is: can I do something about it or just accept the fact that this battery cant last longer? I haven't done battery calibration in the software. All the parameters are default. Maybe that would help? If so, how to do it? Should i just measure the battery voltage when fully charged and modify the BAT_CALIB_REAL_VOLTAGE line?

Thank you for the answer in advance,
Cheers :))

Candas1, 2020-09-13:
Hi, for calibration you need to change this value and also the next one as mentioned in the comments.

It could be that you got a bad battery. Any chance to try with another one?

kulesiax, 2020-09-13:
Unfortunately no, I don't have another one. I contacted the guy who sold me the hoveboard. Despite it's good shape it is 3 years old. The battery got useless because he probably lied that he charged it every now and then. I have to order a new one. It would be a shame if all this work went for nothing and I spent too much money on this scooter to let it just stand unused and rust. Thank you for your time and anwser.

EmanuelFeru, 2020-09-13:
The battery calibration is absolutely necessary to do if you want to use the full battery capacity. Otherwise, the voltage reading is off and even if there is power left the board doesn't know that and it will turn off.

For calibrating the battery ADC reading have a look in config.h It is very similar to the potentiometer calibration. Basically, is charge the battery, read the ADC value on a serial monitor and the real Voltage with a multimeter. Plug them in config.h.

And lastly, with used hoverboards, there is always a risk that the battery is not that good anymore. But you said the Hoverboard was in good condition so I guess also the battery.

It is normal to have the motors light warm to the touch. My hovercar motors also get light warm to the touch. And I can drive it for 1 hour with stock battery also from a used Hoverboard.

juodumas, 2020-09-13:
@kulesiax 
I think you won't get more than 6km with a new/good 10s2p - 36V, 4Ah hoverboard battery unless you limit current to 7A per wheel (default is 15A per wheel and gives good torque & good max speed).

What is the battery voltage after driving 1.5km with your battery?

You can also test with disabled battery calibration by setting `BAT_DEAD` to a very low value (for example 10), the BMS on the battery should handle low voltage and turn off by itself.

kulesiax, 2020-09-13:
I limited the current to 7A and 9A max, I've calibrated the battery aswell. I'll recharge the battery and then I'll report. Wish me luck :D

kulesiax, 2020-09-13:
So I have tested the scooter with following parameters:
I_MOT_MAX = 5A
I_DC_MAX = 7A
BAT_CALIB_REAL_VOLTAGE  4080 (fully charged battery was 40.8V)
BAT_CALIB_ADC 1536 (this is battery ADC value i got in putty)  

The battery life is significantly better now. I managed to make about 3-3,5 kilometres on full throttle most of the time. I have checked the battery voltage with multimeter before and after riding the scooter and it's the same in putty with a difference of 0.1V both times. What I find strange is that the top speed is still 15km/h and torque is the same even though I changed the config from 15 to 5 amps max. Also, the mainboard cut off the power while there was still 35,5V on the battery (3540-3538 in putty to be exact) although in the config.c BAT_DEAD value is equal to 335 and the battery ADC after it cut off the power is 1335, but it should go down to ~1269. 
So what do you think? The battery is faulty or thats how things should look like? To be honest 3,5km range makes the scooter useless. I have no idea what to do now..

Candas1, 2020-09-13:
If the battery is not good there will be a big voltage drop when pulling high current. So you measured 35.5v but while functioning it's probably lower than that. I bought myself that device and tested my hoverboard batteries https://a.aliexpress.com/_BPW2gM . Some were really in a bad shape.

kulesiax, 2020-09-13:
I've ordered a new battery with 4200mAh capacity but from a certified company in my country, not from aliexpress. In case something is wrong or the maximum range won't get much longer I'll just return it to the store and get my money back. I'm really stubborn, I won't give up that easily. Besides I already spent too much money to drop it like that as I mentioned earlier. It has been a fun project and I learned a lot thanks to it.
When I get a new battery and calibrate it properly I'll post results so others can take notes from my mistakes when they come here seeking help with similar problems.
One last question: how is it that the stock hoverboard I bought can (according to the specs) go from 15 to 20 kilometres on one charge with speed up to 15km/h but after modifying the firmware you said my scooter won't go past 6km even with all the lights, sensors and speakers disconnected, a new battery and the same top speed? It seems odd to me. Especially with current limit as low as 5A. Shouldn't the max distance be at least close to what the manufacturer claims?

As always thank you for answering my questions :))

juodumas, 2020-09-13:
> One last question: how is it that the stock hoverboard I bought can (according to the specs) go from 15 to 20 kilometres on one charge with speed up to 15km/h but after modifying the firmware juodumas said my scooter won't go past 6km even with all the lights, sensors and speakers disconnected, a new battery and the same top speed?

I have never tested a hoverboard, so I have no idea about its range and performance. The only meaningful comparison would be to test the same battery in a hoverboard and then with this firmware, same wheel size.

>  What I find strange is that the top speed is still 15km/h and torque is the same even though I changed the config from 15 to 5 amps max.

You should definitely feel a difference between 15A and 5A. Are you sure you have reflashed the firmware after changing the limit? Which config option did you change?

From our makerspace experience with scooters we could go 12km with 2 hoverboard batteries connected in parallel, 15A current limit and 30kmh max speed.
From my personal experience I can drive 12km with my 16S1P LiFePo4 battery and 20km with my 16S2P LiFePo4 battery.

kulesiax, 2020-09-15:
Yes I'm sure. At first I didn't calibrate the potentiometer the proper way because I didn't have a usb-uart converter. I was just uploading the firmware again and again until I found the sweet spot. Dumb of me I know. Then I realized that uart pins on mainboard are 5V tolerant and I have Arduino Uno stashed somewhere when you all said I have to do battery calibration. I used it as a converter and then I could properly do it. It turned out that the minimum poti value was really close, but the maximum was totally off. The reason my scooter didn't slow down after limiting the current is because it never went full-power before. Torque is pretty much the same tho, I really can't feel any difference. Tomorrow I'll get a new battery. 

Please tell me something. Is it possible to connect my new battery and the old one in parallel? Both are protected with BMS. Once the weaker one discharges, it'll cut off the power and the newer will still be working. Am I wrong there? Also what happens with charging in case I do something like that. Will one charger handle two batteries? It could be an issue.

Also also, I've been wondering if I could make use of digital inputs from the right sensor connector on my mainboard. If I'm happy with the new battery, I would like to make the scooter a little bit faster. Problem is I can't just enable field weakening, rise limit current and let my niece drive it because she'll get herself hurt. I was thinking of adding something like ignition switch instead of the button, so that way I could set digital input to high only with a key. Will code like this work in config.h?
if (button1==1) { 
  #define FIELD_WEAK_ENA        1
  #define I_MOT_MAX        15
  #define I_DC_MAX        17  
}
else {
  #define FIELD_WEAK_ENA  0
  #define I_MOT_MAX        5
  #define I_DC_MAX        7  
}
Is it a good idea to change current limit and enabling field weakening mid-drive?

As always thank you for replying. I know I'm annoying with all these questions but I'm a newbie in coding etc and you guys are obviously geniuses and also really good teachers :D

juodumas, 2020-09-16:
> Is it possible to connect my new battery and the old one in parallel?

Better not do this with batteries that have very different performance. Then you connect stuff in parallel, all connected parts must share the current equally. So the good battery will end up re-charging the bad battery and also driving your scooter. But generally if the batteries are quite similar, then connecting them in parallel works fine and they balance each other. Just need an XT60 splitter.

> Is it a good idea to change current limit and enabling field weakening mid-drive?

I think it's dangerous to change field weakening while driving, but I don't know for sure. It's better to do this as part of power-on (for example set different limits if the button is held during power-on). In fact if you have both ADC inputs connected (ADC1 & ADC2), then @EmanuelFeru has something similar already implemented. Check the comments here:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/master/Src/util.c#L497

kulesiax, 2020-09-17:
Hello guys. I received brand new battery yesterday and calibrated it today. I took it for a spin with max current limit set to 6/8A and I thought it would be much better so I also enabled field weakening with 4A. I drove exactly 2km and the battery died. Same as with the old one - after it stopped, I took the cover off and measured the voltage and there was still 36.2V. It seems that, as you said before, when motors drag the current from the battery the voltage drops a lot even with a new battery. I flashed the mainboard again with field weakening disabled and 5/7A current limit. I haven't tested it on the road yet, but I decided to look up what's happening in PuTTY. As expected, when the motors start spinning, the voltage drops quite a lot even when they spin in air without any load on the scooter. I attach the log and also a firmware code that I use. If anyone could check it and see if I didn't f something up I would really appreciate it.

juodumas, 2020-09-17:
Yes, you can effectively disable BAT_DEAD protection by setting it to a very low value, as I have mentioned above. Battery won't be damaged because it has a BMS circuit inside which will turn it off when the battery reaches 26V (I took this number from a random "36V hoverboard battery" listing at aliexpress).

The ADC value which you use when you do battery calibration could be wrong as well. I have seen a couple of boards that report incorrect voltage.

Btw., do not set current limit above 8A/10A when using a single hoverboard battery. Also consider connecting them in parallel after you log a few cycles with your new battery and get similar distance from batteries with the same firmware.

kulesiax, 2020-09-17:
I've considered connecting them in parallel but will it be safe to charge them with the same 1.5A charger I use now at the same time? Will the only downside be very long charging time?

EmanuelFeru, 2020-09-17:
@kulesiax 2 km sounds very low. Are you sure the motors are working correctly? Is wiring of motor phases correct?
There should be low current flow when the motors are in the air. If not, that is a clear indication that something is not OK with the motors.

kulesiax, 2020-09-18:
@EmanuelFeru now that you mentioned it. The front wheel works like a charm. The rear wheel on the other hand starts with a little delay as if there was a load on it and it's hard for the motor to start spinning. Same when I accelerate a bit - front wheel spins faster than the other one. Difference begins to fade as I accelerate the speed to ~1000 rpm. There is also a difference in motors power. The front wheel is much stronger. When I try to block the scooter the front wheel starts a burnout, and the rear wheel doesn't. I connected the wires according to the colors. Is that a wrong wiring issue or there is something wrong with the motor? Turned out my BMS cuts the power at 35V. I was able to drive exactly 4km with one cigarette break at 3,5km. Max speed 16km/h, average - 14km/h. 15 minutes of driving tops.

EmanuelFeru, 2020-09-18:
This means the front wheels are correctly wired but rear ones not.
Sometimes the colors on wires do not match, either motor phases or hall sensor cable.
First, try swapping the rear motor phases in different combination and see if you have the same behavior as in the front.

kulesiax, 2020-09-18:
As I mentioned above swapping phases only got things worse. I've noticed that hall sensor cables on the board are not the same for both motors. While standing in the front of flat side of the board, the colors of hall sensor cables are: black, blue green, yellow, red, but on the other side it's red blue green yellow black. Colors in plugs match. Could that be an issue? I'll try to plug left motor to right side of the board and the other way around and see what happens.

EmanuelFeru, 2020-09-18:
I think the issue is in the hall sensors then. Can you put the wires in the same order as in the front?
You can fix it on the connector side by extracting the pins and putting them in correct order.

EmanuelFeru, 2020-09-18:
Also you can just temporarily use the rear motor with the front board, to check that the motor is OK.

kulesiax, 2020-09-18:
That's exactly what I did. Cables are fine. There is something wrong with the motor. I disassemled it to add a braking disc. Maybe I did something wrong while assembling. I will check it once i have some spare time. Again, thanks for helping me with this :)

EmanuelFeru, 2020-09-18:
Maybe during the brake installation the windings were touched by mistake, creating some short circuit in the winding.. Could be

kulesiax, 2020-09-18:
I was very cautious while doing it. Nevertheless, I'll disassemble it again and check what's wrong. I was thinking of getting rid of the brake disc anyway because it's a little too big for the wheel and the scooter isn't that fast for it to be absolutely neccesary. Electronic brake with slight force will do the job just right. Cheers

closed 2020-10-21 by EmanuelFeru