EFeru/hoverboard-firmware-hack-FOC #99: enlarge the window to enter the torque and speed setting mode

zulkep, 2020-09-27:
Thank you for the firmware, it helped a lot, but there is one question, how to increase the time to enter the mode, adjust the torque and speed, a very short time, and I do not have time to quickly and shortly press the button to turn on the setting mode, how to increase to 2 seconds?

zulkep, 2020-10-01:
what needs to be changed in util.c
to increase the time for a double tap?
/* =========================== Poweroff Functions =========================== */

void poweroff(void) {
	buzzerPattern = 0;
	enable = 0;
	consoleLog("-- Motors disabled --\r\n");
	for (int i = 0; i < 8; i++) {
		buzzerFreq = (uint8_t)i;
		HAL_Delay(100);
	}
  saveConfig();
	HAL_GPIO_WritePin(OFF_PORT, OFF_PIN, GPIO_PIN_RESET);
	while(1) {}
}

void poweroffPressCheck(void) {
	#if defined(CONTROL_ADC)
    if(HAL_GPIO_ReadPin(BUTTON_PORT, BUTTON_PIN)) {
      enable = 0;
      uint16_t cnt_press = 0;
      while(HAL_GPIO_ReadPin(BUTTON_PORT, BUTTON_PIN)) {
        HAL_Delay(10);
        if (cnt_press++ == 5 * 100) { shortBeep(5); }          
      }
      if (cnt_press >= 5 * 100) {                         // Check if press is more than 5 sec
        HAL_Delay(300);        
        if (HAL_GPIO_ReadPin(BUTTON_PORT, BUTTON_PIN)) {  // Double press: Adjust Max Current, Max Speed
          while(HAL_GPIO_ReadPin(BUTTON_PORT, BUTTON_PIN)) { HAL_Delay(10); }  
          longBeep(8);
          updateCurSpdLim();
          shortBeep(5);
        } else {                                          // Long press: Calibrate ADC Limits
          longBeep(16); 
          adcCalibLim();
          shortBeep(5);
        }
      } else {                                            // Short press: power off
        poweroff();

EmanuelFeru, 2020-10-02:
Double tap is for the pedal to engage reverse in Hovercar variant.
The power button long press is for ADC auto calibration or adjustment of max current, max speed.

Which one you mean?

zulkep, 2020-10-03:
>Пятница, 2 октября 2020, 23:59 +03:00 от EmanuelFeru <notifications@github.com>:
> 
> 
>Double tap is for the pedal to engage reverse in Hovercar variant.
how to disable it?, so as not to accidentally press
>The power button long press is for ADC auto calibration or adjustment of max current, max speed.
>Which one you mean?
yes adjustment of max current, max speed.
it is very difficult to catch this moment in order to enter it
almost always you just do not have time to quickly press

EmanuelFeru, 2020-10-03:
The pedal double tap is protected.. It will automatically not engage if motors are spinning. So there is no risk.

For the other one you can increase the HAL_Delay (300) in the poweroffPressCheck() function.

zulkep, 2020-10-03:
one more note, if you decrease the torque, then the torque of the braking function, the same decreases, so much that the brake almost does not work
also, while the wheels are spinning, the power supply to the board does not turn off, in the event of a low battery level, which dangerously drains the battery if it is without a BMS board

zulkep, 2020-10-03:
tell me which file is the multipleTapDetection () variable in?

zulkep, 2020-10-03:
> For the other one you can increase the HAL_Delay (300) in the poweroffPressCheck() function.
I increased the time, but it is still a big problem to enter the menu, it does not work the first time
watch the video [https://cloud.mail.ru/public/ZpGP/5HZZyViAi](url)

EmanuelFeru, 2020-10-21:
@zulkep for some reason I cannot see the movie. Can you share it in a different way?

zulkep, 2020-10-21:
to see the video, you need to copy the link into the browser
another problem, the calibration of the handles does not work (the calibration values are not saved), I compile from the site and in PlatformIO, VARIANT_HOVERCAR

EmanuelFeru, 2020-10-21:
The link does not work for me.

zulkep, 2020-10-22:
https://drive.google.com/file/d/1WjgYfBTcIvcq-5qqzIZWFNdL2MF7W802/view?usp=sharing

zulkep, 2020-11-12:
when the torque is reduced by the handle, the braking torque also decreases, if the torque is minimal, then there is almost no brake, how to make one wheel brake

EmanuelFeru, 2020-12-31:
Last commits should fix the issues in this topic.

closed 2020-12-31 by EmanuelFeru