EFeru/hoverboard-firmware-hack-FOC #103: Hovercar documentation

adverst, 2020-10-02:
Hi - just looking for some clarification on the hovercar variant. The video and other docs you have are all very helpful...

Readme has:
Most robust way for input is to use the ADC and potis. It works well even on 1m unshielded cable. Solder ~100k Ohm resistors between ADC-inputs and gnd directly on the mainboard. Use potis as pullups to 3.3V.

What's the purpose of the 100k resistors to grand on the ADC inputs? It's required on both?

The potis - I think I saw somewhere you can use the 3.3v from the SWD programming, is this right?

The pedals you recommended... I've seen some listed as hall effect, others just as potis, does it matter? So long as they output 0-3.3v depending on position is that all that's needed?

Also with the sideboards, if we connect to ADC1 and ADC2 to the potis how is the left sideboard connected? Does it just need power but not signal?

EmanuelFeru, 2020-10-02:
Hi,
Don't use pull ups resistors, use pull down to GND. Pull downs resistors are to make sure if potis get disconnected (wire broken for example) the motors won't go crazy because the wire is floating.
With pull down, the input goes to a safe 0 when wire is interrupted.

Yes, potis use 3.3V from the programming port.

Both pedals should work, hall based or classic potentiometer.

Sideboard will be connected to the right cable (short wire one) exactly the same way it is connected in the original Hoverboard. Sideboard needs to be flashed with the hacked sideboard firmware.

adverst, 2020-10-02:
Awesome thanks. So only one side board can be used/active if we're using the hovercar variation?

Also the resistor, you're saying solder a 100k resistor between ADC1 and ground and ADC2 and ground?

Everything seems to be working for me, I hooked up a PS3 style thumb joystick with the ADC variation, X to ADC2 Y to ADC1. Can control steer/speed. Will order some pedals and switch over to hovercar variation. Once I'm sorted I can try to commit some additional documentation/steps to help people get going.

EmanuelFeru, 2020-10-03:
Sounds good. If you can help in documenting better that always helps.
Regarding the resistor soldering you are right. And yes, only one sideboard is active to drive the LEDs showing battery level and eventual errors and the photo switches which become buttons to change control mode and activate/deactivate field weakening.

EmanuelFeru, 2020-10-21:
@Candas1 maybe we can do something here, maybe a wiki page.

EmanuelFeru, 2020-12-31:
See https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/Variant-HOVERCAR

closed 2020-12-31 by EmanuelFeru