EFeru/hoverboard-firmware-hack-FOC #104: Motors spinning randomly after a few tries

farev, 2020-10-10:
Hello,

I made a car controlled with a joystick, I use an arduino to communicate with the hoverboard motherboard. It works perfectly when I first try it after flashing the motherboard, but when I turn it off and on again the motors start spinning backwards randomly and when I reflash it again it works fine for 1 try. I have checked and tried with mutiple arduinos and different joysticks and the problem is still there. I don't know if it is a physical problem with the motherboard or an code problem.

annoyinganime, 2020-10-11:
Are you using 15v lines to power the joystick?

farev, 2020-10-12:
The joystick is connected to the arduino, I am powering it with the gnd and 5v pins of the arduino. I have used it for other projects like that and it works perfectly.

annoyinganime, 2020-10-12:
Try cutting red wire (15v) completely from the board - it gives a lot of interference, and that may be an issue

farev, 2020-10-12:
Both 15v wires?

annoyinganime, 2020-10-12:
At least the one that is going along your signal wires

farev, 2020-10-12:
Should I be using the short or the long wires?

farev, 2020-10-12:
I tried cutting the 15v wire and I still have the problem.

Candas1, 2020-10-12:
Are you using USART to control the board ? 
Have you changed anything in config.h ?

farev, 2020-10-12:
Yes I am USART with the tx and rx pins. I have checked several times and I am pretty sure I have everything set up correctly.

Candas1, 2020-10-12:
Are you sure about the values you send from arduino ? what are you getting as Serial feedback ?

farev, 2020-10-12:
Yes I have checked my code several times and what is wierd is that it works perfectly fine the first time I try it after flashing the motherboard but after the first try this problem happens. It works fine, it just doesn't stay still when it should stay still, when I a not even sending anything from the arduino.

Candas1, 2020-10-12:
But I would be curious what are the values of cmd1 and cmd2.

farev, 2020-10-12:
I get this.
[image]

Candas1, 2020-10-13:
Usart is using a start frame and checksum, so those skipped lines are probably due interferences.
The same mechanism is used when sending your commands from the arduino, so this cannot explain your problems, bad messages should be just skipped

farev, 2020-10-13:
So the problem is interferance?

Candas1, 2020-10-13:
No I mean this mechanism makes sure only good commands are handled on the board.

Candas1, 2020-10-13:
I wouldn't expect that the wheels are moving is cmd1 and cmd2 are 0.
And with UART, interferences shouldn't alter the commands.
Bad mosfets could lead to unwanted wheels movements but I don't think it's the case for you.
Do you have a video that can show what's happening ? 
You could share it in the telegram group https://t.me/joinchat/BHWO_RKu2LT5ZxEkvUB8uw

farev, 2020-10-13:
Here is the link to the video: [https://drive.google.com/file/d/19tQyk3IvyBmjHhRzHUEjJwyUGBudF5BH/view?usp=sharing](url)

Candas1, 2020-10-14:
This could happen if the joystick is not perfectly centered. You can try to implement a deadband on arduino. Why are you using an arduino and not connecting the joystick to the board directly?

farev, 2020-10-14:
Yes in my code I have it so that the speed variable is still 0 until it passes the position 100 and -100 for backwards. At first I did use the joystick connected directly to the board but it gave me a lot of more problems, I couldn't callibrate it correctly and it didn't run as smoothly as with the arduino.

Candas1, 2020-10-14:
so even having the deadband implemented you still have those movements ? what if you increase the values ?

If you still want to give a try with direct joystick connection, you can proceed like that:
- use VARIANT_ADC
- uncomment those lines for adc1 and adc2 (https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/8369e4d47dc9594615dd3e2c99550d25a029d944/Inc/config.h#L257) as you use mid resting pots
- Use auto-calibration procedure https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/8369e4d47dc9594615dd3e2c99550d25a029d944/Src/util.c#L429

farev, 2020-10-14:
I will try extending the backwards deadband because the motors are only moving backwards now.

farev, 2020-10-14:
I tried and it didn't work, it seems like something is sneding a wrong value. The two motors move synchronously so I think that something is sending a wrong value.

EmanuelFeru, 2020-10-21:
@farev  Did you the issue?

farev, 2020-10-21:
No, I still have this issue.

farev, 2020-10-24:
Anyone know what may be happenning or how can I solve it?

4hm4d123, 2020-11-07:
maybe the way is to replace stm32

farev, 2020-11-07:
What do you mean with replace stm32?

NickLD, 2020-11-21:
@farev Are you able to paste the snippet of code you are using to talk to the mainboard?

You may want to try and flush the USART on the Arduino when you receive invalid data (checksum fail, then empty the read buffer/stream), as it sorta sounds like the Arduino isn't capturing the frame properly causing every subsequent one to also fail as it's now 'missaligned' .

farev, 2020-11-22:
I am using this code for the arduino. [https://github.com/farev/Arduino-Code-for-Hoverboard-FOC-control](url) I slightly modified the one that is given.

EmanuelFeru, 2020-12-31:
Did you solve it?

farev, 2020-12-31:
I just changed to controling it with a nunchuck, I wasn't actually able to use the arduino. I haven't tried the newest version of the firmware so I will try that, but it is working great with the nunchuk so I think we can close the issue.

EmanuelFeru, 2020-12-31:
Good to hear. Thanks for quick reply and happy new year :wink:

closed 2020-12-31 by EmanuelFeru