EFeru/hoverboard-firmware-hack-FOC #105: Compilation warnings with N_MOTOR_MAX>4095, and top speed is limited

annoyinganime, 2020-10-11:
I was trying to make max RPM value so big that it becomes unreachable, however with values above 4095 I get warning:
>In file included from Inc/defines.h:27,
>                 from Src/util.c:24:
>Src/util.c: In function 'BLDC_Init':
>Inc/config.h:150:25: warning: overflow in conversion from 'int' to 'int16_T' {aka 'short int'} changes value from '65536' to '0' [Woverflow]
>  150 | #define N_MOT_MAX       4096            // [rpm] Maximum motor speed limit
>Src/util.c:218:35: note: in expansion of macro 'N_MOT_MAX'
>  218 |   rtP_Left.n_max                = N_MOT_MAX << 4;                       // fixdt(1,16,4)

If my calculations are correct, that gives me a theoretical maximum speed of 31km/h, which is what I see on GPS logger. But with the same exact setup, and only ESCs changed to VESCs (same current limiting too) I get 43km/h top speed. 
So, is it possible to disable an RPM limit completely?

Candas1, 2020-10-12:
I think your calculation is wrong, 4000rpm is much more than 31km/h, you won't reach 4000 rpm with those wheels. Please check this thread, there are hints how to go faster https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/issues/86

EmanuelFeru, 2020-10-21:
I think explanation from @Candas1  is clear enough. If still issues you can open a new Item.

closed 2020-10-21 by EmanuelFeru