EFeru/hoverboard-firmware-hack-FOC #106: VARIANT_SKATEBOARD and VARIANT_PWM

rodracruz, 2020-10-12:
hello, i have this problem with the hoverboard when installing VARIANTs with pwm signals (https://youtu.be/I_INhBaeQ08 ), maybe is de FOC that is causing problems, i don't know if is already enable or i have to activate some way.  I tried benjaf sk8 version of the firmework (https://youtu.be/nKo6hYb2jNk ) but seems a bit to slow, maybe is the remote controller, and seems a bit erratic, maybe because isn't with weight.

thanks

annoyinganime, 2020-10-12:
Try adding a ceramic capacitor on the board between 15v line and GND. I was having almost the same issue, and that helped me

Candas1, 2020-10-12:
I haven't used PWM but with PPM having a shorter cable would reduce the noise, might also be helpful

rodracruz, 2020-10-13:
> I haven't used PWM but with PPM having a shorter cable would reduce the noise, might also be helpful

That helped a lot with benjaf versión of the firmaware, still i thing is slow compare with the videos y have seen in YouTube but now is smooth and reliable.
Thanks a lot

rodracruz, 2020-10-13:
> Try adding a ceramic capacitor on the board between 15v line and GND. I was having almost the same issue, and that helped me

For the PWM receiver i use de 5v line in the sensor hall, the pwm signal (blue cable) in the signal hall and GND signal (black cable) in the signal hall.
So,  with 15v line do you refere?

EmanuelFeru, 2020-10-21:
@rodracruz Did you manage to fix the jumps in your video?

lukis01, 2021-03-29:
> 
> 
> > I haven't used PWM but with PPM having a shorter cable would reduce the noise, might also be helpful
> 
> That helped a lot with benjaf versión of the firmaware, still i thing is slow compare with the videos y have seen in YouTube but now is smooth and reliable.
> Thanks a lot

I had the same issue, but coverd it by connecting 4 wires at total between board and RC receiver. Firstly I followed this [guide ](https://www.youtube.com/watch?v=2Zg-sp9BD_g) but guy forgot to mention, you have to wire throttle and steering as well. If you wire only throttle, this is what you get :/

closed 2021-07-24 by EmanuelFeru