EFeru/hoverboard-firmware-hack-FOC #108: Scaling speed with hovercar/ADC

adverst, 2020-10-21:
I bought some foot pedals basically teh same as the ones linked from the hovercard video (similar to [these](https://www.banggood.com/Foot-Pedals-Throttle-Speed-Control-Accelerator-For-Golf-Carts-Go-Karts-ATV-Quad-Electric-Scooter-Pit-Bike-p-1460411.html?p=3A291011704562015019&custlinkid=864468&cur_warehouse=CN)).

I went through the calibration, but it seemed like the wheels went to max speed very quickly, the only variance was in the very top end of the accelerator pedal. As in, if you push a tiny bit the motors were slow, push a bit more through all the way down motors were top speed.

I tweaked the ADC calibration to output the max/min for ADC1 and 2. Both accelerators were similar, min was ~1074 max was ~3195.

I also checked when just pushing the pedals a very small amount and it was around 2165. So it means that the first say 5% of the pedal press raises the value to ~2165, then the remaining 95% of the pedals range increases up to 3195. In practice it means that the there are basically two speeds, stop and full speed, if you press the pedal more than a few %.

0 press - 1074
5% press - 2165
100% press - 3195

So what I really need to happen is to non linearly scale the speed between this 5% value and full press.  

Before I go digging to figure out where to do this, any suggestions?

Candas1, 2020-10-22:
Did you use the latest code?

EmanuelFeru, 2020-10-23:
Did you test will wheels off the ground? If yes, it is a normal behavior because Hovercar uses torque mode and if wheels are off the ground the wheels will spin up to Max because there is no load. With wheels on the ground everything should be fine.

Candas1, 2020-10-23:
I had the same thought and it's true torque is not linear when wheels are off the ground.
But adverst seems to say input values are not linear so I had a doubt.

adverst, 2020-10-25:
> Did you test will wheels off the ground?

Ah right, sorry I did. I saw this commented somewhere else, so there is a chance this is the cause. I did try just hand resistance on the wheels to no effect. I haven't gotten anything fully assembled yet so will retest once the wheels are actually trying something

closed 2021-07-24 by EmanuelFeru