EFeru/hoverboard-firmware-hack-FOC #110: IBUS for flsysky

johnnybreizh, 2020-10-30:
Hi there. Thanks a lot for giving us the opportunities to test and work with that great code. 
I have best result with IBUS for flysky ... But just a little probleme.
Forward and backward work very fine.. very great and powerfull...
But when i use right and left command, the wheels turn "very" slowly (divided by 2 ?)  in comparaison of the forward and backward mode.
So how can i change the code to have, in IBUS mode, right and left as powerfull as other commmand ?
Hope my explications are clear because i am not english.. sorry for that.
Have great day at all !

Candas1, 2020-10-30:
You should change default_steer_coefficient in config.h.
It's half the speed coefficient by default

johnnybreizh, 2020-10-31:
Hey ! thanks a lot for your quick answer... I will test that as soon as possible.
It seem to be clear like that.
Great job again and thank you :)

closed 2020-10-31 by johnnybreizh