EFeru/hoverboard-firmware-hack-FOC #112: Reset Button

farev, 2020-11-16:
Hello, is there a way of programming a reset button into the board? Like pressing the on button for 10 sec and the code runs from the beggining again. My project only works one time after I flash the motherboard, each time I want it to work properly I need to connect it to my computer again and flash the firmware. I think a reset button might fix this problem.

NickLD, 2020-11-17:
Sounds like you have a more serious problem at hand, having a 'reset button' isn't a very good patch..
Are you saying you have to re-flash the hoverboard mainboard firmware again, or your external microcontroller?

farev, 2020-11-17:
The hoverboard mainboard firmware. The problem I have is that after I turn off and on again the hoverboard after reflashing it the motors don't stay still anymore. But the first time I try it after reflashing the mainboard it does work fine.

EmanuelFeru, 2020-11-18:
I never heard about such an issue. 
What kind of chip do you have? 
Try to full erase your chip, flash and recheck?
Try another main board to see if that one is OK and rule out a programming issue.

farev, 2020-11-18:
I am using the normal hoverboard chip, and how do I fully erase my chip? I think I accidentally did that one time when I was flashing it and that might have started the problem. I don't have another board to test, I had one but it burned because of a bad battery.

Candas1, 2020-11-18:
In same menu as for unlocking the chip, you have the option to erase it.
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/How-to-Unlock-MCU-Flash

As chip will be empty you will have to keep on/off button pressed while flashing.

farev, 2020-11-18:
I will try it.

EmanuelFeru, 2020-11-18:
Most of us flash it directly from Platformio.

farev, 2020-11-19:
I don't know how to do it from there, but is it ok if I flash it from the stm32 platform or from terminal?

EmanuelFeru, 2020-11-19:
> I don't know how to do it from there, but is it ok if I flash it from the stm32 platform or from terminal?

I only flashed with Platformio and Keil uVision, so I cannot comment. But I know a lot of people are also flashing using STM Link Utility which works as well.

farev, 2020-11-19:
Ok I will try with both, to see if it affects.

farev, 2020-12-09:
Sorry it took me a while but I tested both ways and I still have the problem. I tried doing a complete chip erase and then reflashing it and it didn't change anything.

EmanuelFeru, 2020-12-09:
When you can make a video with the behavior you have. It will help understanding the issue.

farev, 2020-12-16:
The arduino code was causing the problem, so I decided to use a nunchuk instead and it works correctly using this variant. The motors do stay still at 0. To which cables should I connect the nunchuk, to the right side or the left side cables? And is there a way of using the buttons of the nunchuk?

farev, 2020-12-16:
And I feel it is a bit slow, it doesn't reach max speed.

Candas1, 2020-12-16:
Nunchuk can only connect to right side.
Speed_coefficient in config.h is lower for nunchuk variant, you can increase it if needed.

farev, 2020-12-16:
How can I implement the buttons, I don't know how. And to which number should I change the speed coefficient if I want it to reach max speed, right now it is at 8192.

Candas1, 2020-12-16:
32767 is the maximum but it would be too fast.
Check the other variants there are examples

farev, 2020-12-16:
About how many km/h would that be?

farev, 2020-12-16:
I tried it and seems like there is interferance but I that is normal at that speed right?

farev, 2020-12-16:
Also is there a way to add like a deadband for the nunchuk joystick, or is that the filter?

Candas1, 2020-12-16:
You can adjust the deadband parameter in config.h. Doing the auto calibration can also help center the joystick. Please also check the wiki. If you don't use pull up resistors and the cable is disconnected, the board will go full speed.

farev, 2020-12-17:
I couldn’t find where to change the deadband parameter is it the filter value that is with the speed and steer coefficients, and I couldn’t find the wiki if you could please share the link.

Candas1, 2020-12-17:
You have to do little efforts. Please use the search button

farev, 2020-12-17:
I alredy cheked out the wiki but I am still not able to find the deadban parameter in config.h. I found it but not for the variant nunchuk, I found it for the PPM and PWM.

Candas1, 2020-12-17:
Then it means you are not using the latest version of the firmware

farev, 2020-12-17:
Yes I am not using the latest version of the firmware, I will update it. And how can I make the buttons work like a throttle and smooth brake?

Candas1, 2020-12-18:
Hi, it's not something you can just activate with a parameter.

farev, 2020-12-18:
I know but can you like direct me to where I can do it, or atleast help me activate the inputs from the buttons.

farev, 2021-02-11:
So I was able to control it with a nunchuk but then I updated to the newest frimware and it doesn't read the nunchuk.

closed 2021-02-13 by farev