EFeru/hoverboard-firmware-hack-FOC #116: 3 beep error message

Istria1704, 2020-12-09:
Hello,

I am still trying to figure out how the hoverboard hacking works. 
I got the original hacked firmware working in demo mode. The wheels sweep back and forth gently. 

But I'm having trouble controlling it from my Arduino. I tried both a Wemos D1 Mini and a MD328 arduino-compatible mcu set at 3.3v.

I almost dreamt about the 3 beeps last night... 

I connected GND, TX and RX to the Arduino devices. Also tried swapping TX and RX cables around. And tried both UART2 and 3.

I see in the example arduino code the baud rate to the hoverboard is set at 38400. Is this correct? I tried 115200 also, but with no joy. 

Any suggestions?

Istria1704, 2020-12-09:
I tinkered some more with it. Trying ADC mode.

When I feed it between 0 and 1V on the multimeter, it is silent (no beeps). When I pass 1V, it starts beeping with 2 beeps (adc timeout). When going below 1V again, the beeping stops. 

I've gone up to 1,5V, but still no movement.

EmanuelFeru, 2020-12-09:
Hi,

Relecently i changed the code to 115200 baud rate, but indeed I forgot to change the Arduino code. But if you change that it should work.

Regarding the ADC variant that should work, but requires a calibration first.

Istria1704, 2020-12-09:
Thanks. I have gotten adc to work now with the original hack firmware of the GPT18 guys. I will try some more with your firmware tomorrow when I make a cleaner testsetup.

Istria1704, 2020-12-10:
Great success, I got it to shut up (the 3 beeps). =)

Using a Wemos D1 Mini the 3 beeps stop as soon as I connect the Wemos TX to the Hoverboard.

I left everything in the example script standard, except for the HoverSerial baudrate, which I changed to 115200. And the Softwareserial pins, which I set to D6 and D7.

At the RX side, the following datastream is coming in.

[image]

Still no motor movement though. I will look into trying that cmd2 override. In which script should I do that?

Thanks for the help in advance.

EmanuelFeru, 2020-12-10:
Good to hear the serial is ok.
For some reason I see the data sent from Wemos is 0. Fields 1: .. and 2:.. so it won't move. Is there a way to check with a Serial Monitor that the values in Wemos are changing?

>> Still no motor movement though. I will look into trying that cmd2 override. In which script should I do that?

Insert the lines below [here in main.c](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/3fcae4fc8bb60bc6d26a7466a681183c1965cad1/Src/main.c#L257-L258)
```
enable = 1;
cmd2 = 200;
```

Istria1704, 2020-12-10:
The motors do move now, commanded from the arduino. The problem seemed to be I didn't to the autocalibration. As soon as I ended the autocalibration it started up with a huge kick (because the arduino was at almost max power in its cycle). After that it continued to work as intended. 

I don't remember of the cmd1 and cmd2 values were != 0. I will check later.

Istria1704, 2020-12-10:
Hmm, 1 step forward, 2 steps back.

I adapted the arduino script to Send a 'power' value between 0 and 100 to the hoverboard: Send(0, power);

The motors moved, but 0 was not off. The serial feedback gave cmd2 value of 800ish.

So I thought I'd recalibrate. After I did another auto calibration, it didn't work anymore. The serial feedback gives all zeros. Same as yesterday.

Could you please confirm weather calibration has any function when controlling from UART?

EmanuelFeru, 2020-12-11:
Yes, over thing to keep in mind the auto calibration is active for all variants.
For UART sometimes is not needed. So what you need to do is to go in config.h and set both input1 and input2 Type to 1 (Normal potentiometer).

Then, because you already flagged before, either do a full chip erase or change the FLASH_WRITE_KEY in config.h . These will make sure that any calibration done before is ignored.

Istria1704, 2020-12-12:
I think we can close this issue. I have it working now perfectly.

I think the problems I had in the beginning was that I was using the wrong Wemos IO pins for the softwareserial.
According to: [http://www.esp8266learning.com/wemos-d1-mini-hardware.php](url), There are 10K pullup resistors soldered to D3 and D4. On top of that, the onboard LED is wired to D4. I had not taken that into account. Because at first I tried D2/D3 for RX/TX. And after that D4/D5, I always had a 'pulled up' pin.

I'm still not sure why the MD328 gave problems. All I can conclude for now is that the softwareserial TX didn't work on D3, but worked on another pin.

I do have 1 final question. I read somewere (I can't find it now anymore) that during autocalibration, the adc inputs should be held for a while at their endpoints. Because otherwise the filters(?) would not properly read or process them.
In the arduino example script, the commands move quite quickly from min to max. Could this maybe have been the problem I had using autocalibration in UART mode?

Also 1 small suggestion for the arduino script: 
Instead of:
`int iTest = 0;`
This would make the controller startup at 0 instead of max power. A bit less scary :)
`int iTest = SPEED_MAX_TEST/2;`

Thanks for the help and support thusfar!

EmanuelFeru, 2020-12-12:
Good to hear is working.

And yes during autocalibration is good to hold the ADCs at their endpoints for the filters to properly converge. 1 or 2 seconds should be sufficient.

Now, yes there is a problem if you run the Arduino code while the auto calibration is in progress. Because no matter what SPEED_MAX_TEST you set in Arduino it will be scaled up to max Motor potential. 
I set in Arduino #define SPEED_MAX_TEST 300 and that should be a safe value but because you ran auto calibration the 300 is interpreted by mainboard as the max, so it will produce 1000 to the motors.

My advice is to only run the calibration when you use the full range on Arduino side.
If I were you, i wouldn't run autocalibration because the min max limits on Arduino are known, so you can just plug these limits on the mainboard. Then it should be ok.

closed 2020-12-31 by EmanuelFeru

farev, 2023-09-12:
I can't seem to get the 3 beeps to go away when I try using the uart variant. I am trying to connect it to an arduino, I had gotten it working perfectly before but after I reflahed the firmware I can't get it to work. Any ideas how I can get it to work? The only problem I have is that the 3 beeps which means a timeout won't go away even after properly connecting it to the arduino.

Istria1704, 2023-09-12:
Can you share some more info about your setup?

- Which Arduino board? 
- Which pins are you using for serial? 
- How did you wire it exactly? 
- Are you using the (unmodified) example arduino code from the repo?

farev, 2023-09-12:
I am using an arduino nano.
I am using the tx and rx pins of the arduino nano and on the HoverSerial is 2 and 3.
I have tried using the unmodified example arduino code and I still get the problem. 
Also something new that happens this time that didn't happen when I had gotten it working before is that the hoverboard turns on when I connect either of the tx or rx cables while the ground is also connected to the arduino.

Istria1704, 2023-09-13:
There is (probably) your problem: The 2 and 3 in HoverSerial stand for the Arduino pins used. So don't use the RX and TX pins, but use pins 2 and 3.

If it still doesn't work try HoverSerial(3,2) instead of HoverSerial(2,3), in case you have the TX and RX swapped.

Hope it works! 

P.S. Isn't a nano always 5V logic? I'm not sure the hoverboard can tolerate that on it's RX pin. But I'm not sure.

farev, 2023-09-14:
Yes, I was using the wrong pins. I don't know how I didn't see that mistake before. It does work perfectly fine with the arduino nano and the UART on the left side (short cables) can tolerate 5v.

oorischubert, 2024-09-27:
Hey if any of you happen to see this almost a year later, I am also having issues and am sweeping the web for solutions. I followed EEferu's instructions and added the two lines to main and have confirmed that the motors do in fact move, and when on 115200 I receive serial data back on USART3 (albeit with many error lines). The issue is that when I try sending data nothing happens other than the motors becoming rigid. Even more odd is when I specifically send a move instruction via ros2 I do see an uptick in power usage but no movement at all. What in the world could be the issue? It seems that the tx line is being acknowledged but the actual movement command is nit being implemented. Thanks!