EFeru/hoverboard-firmware-hack-FOC #117: Motor control for escooter (based on Xiaomi M365)

Koxx3, 2020-12-10:
Hello !

First, I want to congratulate all members participating for this project. It's really awesome.
I am working on different projects to make open-source electric scooters : https://github.com/Koxx3/minimo/blob/master/README.md

I am currently trying to improve the behavior of existing controllers... but with your code, I am wondering if **I could create my own electric motor controller.**

My goal is to use up to 20S batteries (84V fully charged), so I know I'll need a strong hardware for that. But I'll keep that for later.

For now, I'd like to replace a Xiaomi M365 firmware. Why this hardware ?
It use a STM32F103C8 microcontroller, I has a strong power electronic and can be easily customized (talking about the hardware part).

What inside a Xiaomi controller board beside the microcontroller ?
* 3 low side current shunt (with LMV324 amplifier) 
* 3 resistance voltage dividers for over-intensity protection on each motor phase
* 3 independant MOSFET drivers
* 6 MOSFETs
* 6 PWM directly connected to the drivers.
* temperature sensors and few other things
* a voltage divider for battery level mesurment
* a good disipator (at least enough for the 350W original motor)
* 1 half-duplex UART and a 1 full duplex UART

the motor has 3 hall sensors

What I would like to achieve : **a torque and velocity controlled FOC loop with this hardware.**

If I can do that, I can try to build a strong electronic to deliver the max power I would like to have.

So ...  I'd like to know what is currently missing to achieve my goal.

For all the "logic" part, I have no trouble developing it.

What is missing for my project ? What should I take care while porting this on Xiaomi hardware ?

Thanks again.

FYI : Here is the retro-engineering schematics on this controller...
    https://github.com/Koxx3/minimo/blob/master/documents/Driver%20V2.1%20M365.pdf

Foujiwara, 2020-12-15:
Thanks for your feedback.
It now works on the Xiaomi M365 controller ;)
We'll have to tune the whole parameters to get a good control.

Candas1, 2020-12-15:
OK n'hésitez pas à partager vos exploits ;)

Koxx3, 2020-12-15:
ahahah. tu es francais ?

Koxx3, 2020-12-17:
finally restarted from scartch (with the latest release) :
https://github.com/Koxx3/hoverboard-firmware-hack-FOC/tree/M365

but :
- ported to Stm32CubeIde
- adapted for stm32f103c8t6
- adapted pinout for xiaomi m365

commutation & sine working.
foc : ko

any idea what it could be ?

Koxx3, 2020-12-17:
FOC is working now.
It was because of bad reading on ADC1&2.
No sure I understand how ADC & DMA works. I'll have to read the STM documentation ;)

EmanuelFeru, 2020-12-18:
Great! Nice to hear.

closed 2021-07-24 by EmanuelFeru