EFeru/hoverboard-firmware-hack-FOC #118: new (repaired) board shuts down at about curR_DC =5 amps :-/

RoboDurden, 2020-12-11:
Hi Emanuel, greetings from a cold winter journey with a broken control board that i have replaced with another one i had repaired..
Motors are running as they should but at about 5 amps (curR_DC from bldc.c), the board shuts down.

I_MOT_MAX , I_DC_MAX , FIELD_WEAK_MAX have no effect.

Maybe it is only when my 7s battery is fully charged !

When the board shuts down, my serial feedback no longer is received so i guess that the main loop has been cancled.
After turning the board off and on again, the serial feedback returns.

I am using an older fork of yours, so this might no apply to your current version.

But you might have an idea if i have a hardware problem or a software problem.

Any(!) Ideas welcome,
the Roland :-)

RoboDurden, 2020-12-11:
If I set CTRL_TYP_SEL    0 = Commutation, the board is more stable. Yet when my arduino shunt reading exceeds about 10 A the controller again shuts off.
Did not have my serial feedback reardings but 2 motors at 5 A sum up to 10A.
The power at the wheels is as it should be. So I would like to disable a software shutdown.
In three days driving I would have a replacement board.
Pulling my solar car will rather take a week.

I don't think there is a hardware shutdown on the circuit board.
And it is not a reset!

EmanuelFeru, 2020-12-11:
Hi

The shutting down is a sign of worn  battery. Because as you draw current from the battery the voltage drops that much that either the main board detects too low voltage  and turns off or the BMS itself shuts down.

RoboDurden, 2020-12-12:
YES !!! you made my day.
It was the `poweroff()` inside the util.c :

```
void poweroffPressCheck(void) {
/*...*/
        } else {                                            // Short press: power off
          poweroff();
        }
      }
```

I guess the HF noise on my power line increased with more power on the motors that cause the power button to be triggered..
Maybe you want to replace that `else`  with a `if (cnt_press >= 10) // > 100 ms`

As i don't need the `poweroffPressCheck `i have uncommented it in my main.c :-)

I have two 700 Wh 7s10p 100 cycles batteries and one 500 Wh 7s7p 20 cycles battery and every battery  has it's own bms.
24 Volt is absolutely okay for these boards and match my solar panel, my chain saw, my pedelecs and my house system voltage.

Many thanks again !

P.S. have you by now added a max speed for left and right wheel so the speed mode works properly in curves ?
That is why i am still sticking with my old fork of your nice FOC firmware.
Yes i preferred my crc32 but as you said you improved your serial communication i guess i would like to give it a second try.

EmanuelFeru, 2020-12-13:
Debounce on the power button for sure won't hurt. I will take a look. thanks for the input.

Regarding the speed difference Right-Left in curves, if you use Speed mode, you neeed to manage your self via the `steering` input signal. And the UART, I am not sure what version you use but it has been improved indeed.

RoboDurden, 2020-12-15:
So you still neglect that a car has no steering input but a steering wheel :-(
Did you really attach a potentiometer to the steering wheel of that fun-project toy cart ????

EmanuelFeru, 2020-12-15:
I use torque mode so no need for it. In speed mode, how else can you do it without knowing your steering angle?... You need some sort of sensor.

RoboDurden, 2020-12-16:
Maybe you want to reopen https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/issues/68 :-)

You don't need a steering angle. A max speed for left and right will do. Look at the Variant_uartcar of my fork: 

https://github.com/RoboDurden/hoverboard-firmware-hack-FOC/blob/3fcb0859f8ec1ab91fd04454f7f83d4aa6217c27/Src/main.c#L279

I needed to add my own closed loop in main.c with your voltage mode :-(
I would rather like to you use your speed mode.
But it is unusable for cars (no steering input).

closed 2021-07-24 by EmanuelFeru

AlaaImter, 2022-04-03:
Hello
I am wrong and wire the potentiometer with the 15v wire and the board shut down, can i repair it ?

RoboDurden, 2022-04-03:
Unlikely.
You should have watched my tutorial first: https://www.youtube.com/watch?v=MJxzChkNDxQ "Hoverboard flashing for beginners (red != +14V)(adc wheelchair)"
Now you can watch my https://www.youtube.com/watch?v=0o2TOr6Tpg8 "part 2: hoverboard repair (+ let us make an expert system)" and check the two voltage regulators..
If there is hope, do not post here but under my (red != +14V) video.

Candas1, 2022-04-03:
Why not here also? 😅

RoboDurden, 2022-04-03:
Because this issue is about "board shuts down at about curR_DC =5 amps".
"First rule: you always speak yes." (Robo Durden)