EFeru/hoverboard-firmware-hack-FOC #122: uart Arduino Nano and hoverserial

cecca9596, 2020-12-25:
hi I'm trying to use uart but with bad result on the USART3.
i have the timeout of the signal with the classic beep.
I tried different pin for tx and rx on Arduino nano with the correct value on hoverserial sketch. but nothing
could be the baudrate too high for the Arduino Nano?
i will try with Arduino mega.

or maybe something on the calibration.

  // #define SIDEBOARD_SERIAL_USART2 0
  //#define CONTROL_SERIAL_USART2  0    
  //#define FEEDBACK_SERIAL_USART2      

  // #define SIDEBOARD_SERIAL_USART3 0
   #define CONTROL_SERIAL_USART3  0   
  #define FEEDBACK_SERIAL_USART3    
 #define PRI_INPUT1             1, -1000, 0, 1000, 0    
  #define PRI_INPUT2             1, -1000, 0, 1000, 0

cecca9596, 2020-12-26:
another thing:

I can read the debug serial with my Jetson Nano but not with the Arduino.

instead I can't read the feedback serial even with the Jetson Nano.

it's pretty frustrating....

cecca9596, 2020-12-26:
we can close I solve the issue

closed 2020-12-26 by cecca9596

Koxx3, 2020-12-26:
what was the issue ?

cecca9596, 2020-12-27:
> what was the issue ? Le sam. 26 déc. 2020 à 17:47, cecca9596 <notifications@github.com> a écrit :
> […](#)

not al the pin on Arduino mega can be used as tx rx, the code on hoverserial use the 2 and 3.

using different pin I solved the problem.