EFeru/hoverboard-firmware-hack-FOC #123: Blutooth module BT05 and an Andriod App And

ricoghardforth, 2020-12-30:
Has anyone developed an android or IOS hoverboard phone Bluetooth app for this custom hoverboard firmware yet.   Couldn't  the spare side port channel when set for debug have the UART/serial coms connected to something like a BT05 Bluetooth module and then to phone so we can have a look at data from the controller like battery volts, current, watts, faults etc.  
Similar to the ones I use with the VESC on my Esk8 build.
Maybe even control the device via the phone.  
This isn't really an issue but I can't find a way just to post a general comment on the software please let me know if I'm supposed to do it some other way thanks.

vamfun, 2020-12-31:
Hook Uart to ESP32 and you have built in Bluetooth or Wi-Fi .   See YouTube projects for sending data to phone.   I've done it with Wi-Fi.   You can set up a server and have data sent there or there are apps that connect like REMOTEXY .

Koxx3, 2020-12-31:
i modified the serial link protocol, but i think you take a lot from my project : https://github.com/Koxx3/SmartController_SmartDisplay_ESP32

Koxx3, 2020-12-31:
[image]
[image]

Koxx3, 2020-12-31:
[image]

Koxx3, 2020-12-31:
[image]

EmanuelFeru, 2020-12-31:
Canadas did amazing stuff in this direction too, check here:
https://candas1.github.io/Hoverboard-Web-Serial-Control/

closed 2021-07-24 by EmanuelFeru

mouixong, 2021-09-12:
@Koxx3 Can the motherboard of the balance car be brushed?