EFeru/hoverboard-firmware-hack-FOC #124: sorry, again: max rpm speed for left and right

RoboDurden, 2021-01-07:
Sorry to ask again :-/ But i still would like to return from my 6 months old fork to your original.

But i need my VARIANT_UartCar where i only set one speed which translates to the max speed of left and right wheel.
(There is no steering input with a car, only a steering wheel.)

your target speed enters BLDC_controller.c in bldc.c here: https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/63276132a7f97457a2bd19a29451e88569920985/Src/bldc.c#L173
`and
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/63276132a7f97457a2bd19a29451e88569920985/Src/bldc.c#L211

with   `rtU_Right.z_ctrlModReq  = ctrlModReq;` this pwmr will be interpretet as rpm (if i remeber correctly)

Now i only need to find the line in BLDC_controller.c where the rpm of the wheel is calculated and replace it with a global variable that i update to be the max of left and right.

But the code lines with `(rtU->z_ctrlModReq == 2)` are unreadable to me :'(

And i also do not find any code where a rpm value is calculated :'(

Do you calculate the right rpm from 

```
    rtU_Right.b_hallA       = hall_ur;
    rtU_Right.b_hallB       = hall_vr;
    rtU_Right.b_hallC       = hall_wr;
```

In that case i could set rtU_Right.. and rtU_Left... with the same values after i somehow have found out which set of hall inputs imply a faster speed.

But as pwmr and pwml are rpm values, you must have a closed loop with the rpm values of the wheels.
But i am too stupid to find them :-(

Grettings from Germany,
roland.

RoboDurden, 2021-01-07:
ah maybe i found the true speed that will be compared against the target speed: https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/63276132a7f97457a2bd19a29451e88569920985/Src/BLDC_controller.c#L1892

Candas1, 2021-01-07:
Have you checked the main loop?
Cmdr and cmdl are calculated there
Why are you not using torque mode?

RoboDurden, 2021-01-07:
thanks Candas1 for your answer. Yet i can not understand why you all do not want to understand that a car only has a speed input and no steering input. This one speed is split into two wheel revs by a differential gear.

I need to limit my solar car to 6 kmh so i need a speed mode, not a torque mode.

Yes, i could use left.n_mot and right.n_mot to detect the car driving in a curve and calculate a steering input steer1.
So a target speed of 1200 rpm would translate into 1000 rpm for the left wheel and 1400 rpm for the right wheel.
this 1200/1400 would lead to an updated steer2 that should already be close to the true difference between the inner wheel and the outer wheel.

But hwat happens when one wheel slips ? Then the logic would interpet it as a curve and would power down the only wheel left with traction. Because it thinks car is rotating with the inner wheel standing still.
Okay i could translate the 1200 rpm into 1200/1400. Meaning that the inner wheel is always the target speed. 

Still i am unsure if this will work at all because i take the different wheel speeds to calculate a steering input which then will lead to new different wheel speeds = a new closed loop on top of the closed loop in BLDC_controller.c - That might get out of control in an unexpected situation on the streets !

If i go my way with target max speed, the slipping wheel would stop at 1200 rpm and the good wheel would also power down.

But if i would set the target speed as min speed, the slipping wheel would go max but the good wheel would be taking all the load to reach 1200 rpm :-)

i want to stay as close to your repo as possible and Emanuel already offers a speed mode.
But his speed mode only works for each wheel independently.
A car has no indepentend wheel left and right. 
It only needs one global variable n_motMax or n_motMin to couple the two independed closed loops.

Why is it so difficult to understand ?

EmanuelFeru, 2021-01-07:
Roland, for me it is still not clear what you want to achieve. You mentioned max speed which you want to plug in as target speed.
Normally a limitation is used to limit/saturate the target and not send as a target itself.

Overall what i understood you want to use speed mode but avoid squeaking wheels in corners. For that you just need to compute the correct speed target left and right and send it to the motors. On top you can limit the speeds by saturating the target speeds.

Candas1, 2021-01-07:
Roland we are just trying to help, maybe there are aspects of the firmware you missed.
Have you tried this parameter to limit the speed ?
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/63276132a7f97457a2bd19a29451e88569920985/Inc/config.h#L150

As I and Emmanuel mentioned, you can calculate whatever cmdL and CmdR command you want in main loop.

RoboDurden, 2021-01-07:
Thanks Emanuel for still answering me :-)
Yes i want to avoid squeaking  wheels in corners.
Yes i could add a potentiometer to my steering wheel and use the geometry of my four wheels to calculate the steering radius :-(
Or i could calculate the steering input from the realtime left.n_mot and right n_mot as i wrote above.
But i doubt that you will allow such a "closed loop on top of a closed loop" in your repo.
Yet a single global variable and one #define SPEED_IS_MAX_SPEED will not hurt :-)

@Candas, N_MOT_MAX is only the limit not a target speed.

EmanuelFeru, 2021-01-07:
I was typing while you were typing :wink:
The cars on the streets don't have an "input speed" they have an input torque. That goes into the differential shaft and you get torque left and torque right. There is no car where you have input speed. Also in cruise control the controller overwrites the pedal request sending a torque request to the engine.

In the same way you can use Torque mode. I added already a control loop to limit the maximum speed, you set that to your equivalent 6 km/h. It it is not intensively tested so if you don't like the behavior feel free to implement your own closed loop speed limitation that will reduce torque target in order not exceed your desired 6 km/h.

RoboDurden, 2021-01-07:
Where is this control loop to limit the maximum speed :-)
Is this a maximum speed a **variable** that i can set in main.c via cmd2 ?

Again i am not happy with a "control loop on top of a control loop" - every engineer will tell you this is very bad style.
It would be okay if these two closed loops would work on different time scales. But when a huge trucks wants to roll over me i have to turn aside in a tenth of a second, which will be on the same time scale as your BLDC_controlerl.c ?

And yes i am very happy to advance a crossing with 1 kmh or go down a very steep descent with 3 kmh. Torque mode is only for fun ;-)

EmanuelFeru, 2021-01-07:
Torque mode is what all vehicles on the streets are using :)
Give a try for Torque Mode or Voltage mode with the maximum speed limitation parameter. If look in the util.c how is this set you can change it during run time and for each motor.

RoboDurden, 2021-01-07:
I don't think a max speed limit will help me.
In a curve, the two speeds have to be different.
If i set a limit to 6 kmh = 1200 rpm, the outer wheel will be limited. But how to set the torque of the inner wheel ?
the two motors can not be independent. A limit for both motors will not couple them.

I am only a physicist, no engenieer. I can only do simple closed loops and have no idea what saturation means.
But i believe for sure, that there is always a target value and a real time value.

Why don't you let me try to set the realtime values of both motors to the max or min of the two realtime values.
Then the two motors will be coupled in only one closed loop, the one you already have in BLDC_controller.c
It already works with an additional primitve closed loop in the main.c of my fork, but you will not add that to your repo.
So i want to find a solution with only your nice speed closed loop in the BLDC_controller.c

Where are the realtime speed values in BLDC_controller.c that are matched against the target speeds pwmr and pwml ??

I think i am already pretty close with Switch2 https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/63276132a7f97457a2bd19a29451e88569920985/Src/BLDC_controller.c#L3143

EmanuelFeru, 2021-01-07:
### Speed Control Mode code snippets
[image]

---
The error between target speed (in `rtb_Saturation`) and measured speed (in `Switch2`) is calculated here:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/63276132a7f97457a2bd19a29451e88569920985/Src/BLDC_controller.c#L2579-L2580
where, the target speed `rtb_Saturation` is set here:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/63276132a7f97457a2bd19a29451e88569920985/Src/BLDC_controller.c#L2570-L2575

---
Finally, this error is fed into the PI controller, as you see below:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/63276132a7f97457a2bd19a29451e88569920985/Src/BLDC_controller.c#L2589-L2594

I hope this helps, because I cannot support further.

RoboDurden, 2021-01-07:
So indeed `Switch2` is the realtime revs value that i need to replace with the max (or min) of left and right.
Will try that tomorrow. I am not very confident as there are so many `Switch2_xy `and i do not see where a simple n_mot is fed into "err" like shown in the snippet image.

But thanks anyway. I did not expect that you would help me at all.
So thank you for your insight and i will not think bad of you when you stay silent from now on :-)

closed 2021-07-24 by EmanuelFeru