EFeru/hoverboard-firmware-hack-FOC #125: Arduino USART3 Not Working

DanielQuintana2, 2021-01-08:
Hello, I am very new to this HoverBoard hacking but I've finally been able to upload the Firmware to the HoverBoard but now I'm having trouble controlling it with the Arduino.

I have been using RoboDurden's online compiler to compile a .bin file which successfully uploads to the mother Board (https://pionierland.de/hoverhack/ found under EmanuelFeru/hoverboard-firmware-hack-FOC). In the Config.h file I've uncommented the #define VARIANT_USART and I followed the instructions in the hoverserial.ino file.

[image]

I have my Arduino connected to the Right set of wires (short) and have used a multimeter to ensure that the Black wire was truly ground. I've checked multiple times to ensure I had the Rx and Tx of each device connected properly as well.

I've uploaded the hoverserial.ino to an Arduino Nano and I currently trying it with an Uno. I've tried different digital Pins for the serial but those seem to have no different results. 

Really, the only thing I can think of is that I'm somehow not configuring the firmware right in the config.h which is why I'm reaching out here. Possibly someone else is has run into similar issues(?) Also, I'm not sure if it's relevant but the Red Wire (Power) on the Right set of wires reads zero volts when I measure it. The black wire is surely ground but perhaps this has something to do with it (I am unsure since it is unnecessary anyway)?

(Also this is my first time using GitHub for issues so I apologize if this is not the proper way to ask questions)
Thank you.

EmanuelFeru, 2021-01-08:
Hi @DanielQuintana2 and welcome! ;-)

- At a first look at your config.h I am not sure how were you able to compile the firmware because normally it should throw an error if you have both of the following defines enabled:
```
#define CONTROL_SERIAL_USART3
....
#define SIDEBOARD_SERIAL_USART3
``` 
You have to choose one because they are on the same cable. You cannot have both an Arduino and a Sideboard connected to the Right cable (UART3).

- Double check that you uncommented the correct Variant in `config.h` if you use the online compiler :
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/63276132a7f97457a2bd19a29451e88569920985/Inc/config.h#L7-L24
Or the right Variant in Platformio if you use Platformio:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/63276132a7f97457a2bd19a29451e88569920985/platformio.ini#L8-L22

- regarding the fact that you don't read any voltage on the Red wire of the UART cables is strange. Is the red LED on the mainboard lit when you measured the voltage and do you hear the turn on melody when you press the power button? The board has a latch mechanism activated by software to close the first TIP127 and give power to the board. So, if the firmware did not flash correctly -> then no latch -> no power to the board.

DanielQuintana2, 2021-01-08:
Hi @EmanuelFeru 
Thank you for your reply! 
Just to double-check, do I need to change anything with the 
`#define CONTROL_SERIAL_USART3
....
#define SIDEBOARD_SERIAL_USART3`
in the config.h?

It looks like I've uncommented the correct Variant:
[image]

As for the voltage, I do get around 15v off of the left one just not the right one (short cable). The LED is lit when I've been doing all of this but I do not hear a melody when the power button is pressed. Perhaps this latch is what is giving me trouble...In order to keep the board on I either have to hold the power button down or use a jumper wire (jumping the power button terminal). When I flash there are no errors but perhaps there is something else wrong?

I've Uploaded Niklas Fauth's firmware onto the motherboard without changing anything and it completed the motor test completely fine but that's about it. Thank you again for your Reply!

EmanuelFeru, 2021-01-08:
- If you were able to spin the motors, that is a good sign... However, on the other side not hearing the melody is not a good sign.. Have you tried other Variant like ADC? Usually that is the lightest in terms of memory.
- I am still puzzled that you only have 15V on one cable...
- For flashing I always did it by holding the button. The advantage is I can release it quickly at any time. 
- Also it would help to know what board you have, STM, GD, a picture can help.

DanielQuintana2, 2021-01-08:
Hi @EmanuelFeru 

I have not tried the ADC variant. I'll try that rn and let you know what happens.
When I flash I guess jumping the terminal is equivalent to holding the button. But is it normal to have to either hold the button or use a jumper in order to keep the board on after I've flashed?
I didn't know there were different types, here is a picture 
[image]
[image]

It's kinda hard to read the main chip but it seems to be a GD.

Thank you.

EmanuelFeru, 2021-01-08:
Oh, you board is different.. it doesn't have the same shape like mine. So i am not sure if the pinout and connections are the same. You can join the Telegram group (link in readme) and ask if somebody had experience with this kind of board.

DanielQuintana2, 2021-01-08:
Ah okay, I'll be sure to check it out. Thank you!!

Candas1, 2021-01-09:
This board has a additional white wire for wheel temperature sensing, and maybe other differences. I think someone was investigating it in the group, maybe they can share their findings.

DanielQuintana2, 2021-01-11:
Hi @Candas1 
Thank you for your comment and for asking on the group. I was actually able to solve it in issue #126 in Niklas Fauth's forum...here's the link incase someone else is looking for it https://github.com/NiklasFauth/hoverboard-firmware-hack/issues/162

Turns out some GPIO pins needed to be changed ^^.

closed 2021-01-11 by DanielQuintana2

ayahelmyy, 2024-11-04:
hi @RolandoQuintana , i came little bit late XD , but i have the same problem with arduino and have the same code that you provided in the pictures , and the 5v pin the right wires is 15v not 5v, i don't kno why , any help ?

RolandoQuintana, 2024-11-04:
> hi @RolandoQuintana , i came little bit late XD , but i have the same problem with arduino and have the same code that you provided in the pictures , and the 5v pin the right wires is 15v not 5v, i don't kno why , any help ?

Hello @ayahelmyy ! It's been so long I honestly don't remember any of this but it looks like following instructions from https://github.com/lucysrausch/hoverboard-firmware-hack/issues/162
solved the problem for me.

ayahelmyy, 2024-11-04:
I saw it already, but it's a different problem not related to UART !

RolandoQuintana, 2024-11-04:
Sorry then, that was the solution for me.

ayahelmyy, 2024-11-04:
Yes i know , i meant can you help me with the solution , the second issue was related to the voltage , not uart, right?

Candas1, 2024-11-05:
The picture on the main page shows that you should use 5v from the hall sensor cable.

The sideboard cable has 15v.

If you have 15v on the hall sensor cable, it's bad. Probably the 5v regular is defective.

ayahelmyy, 2024-11-05:
@Candas1 i don't use 15v with hall sensor , it's connected to its cable , i only use the UART3in the right side , i didn't understand your point of view , please explain

Candas1, 2024-11-05:
https://github.com/EFeru/hoverboard-firmware-hack-FOC#hardware

ayahelmyy, 2024-11-05:
https://www.youtube.com/watch?v=E6JbFnRiQ5g&list=PLXJDPdNEjPS4lyRpuGIr0aqRVPMaLvePO&index=2

Candas1, 2024-11-05:
Check the wiki for troubleshooting
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-USART