EFeru/hoverboard-firmware-hack-FOC #126: Charging enables motors on Gen. motherboard ...

vamfun, 2021-01-09:


Candas1, 2021-01-11:
Hi, do you have a picture of your mainboard you can share ?

PA1 is the ON/OFF button in the firmware. In the changes you share it seems to be used for DCLINK ?

vamfun, 2021-01-11:
@Candas1 ...I added a picture to my original post.  
Thanks for the board spreadsheet.  How do you know all the boards are GD chips rather than STM32?

Candas1, 2021-01-11:
OK your board looks quite common.
But you have 6 wires on hall sensor cables, one being white?

I just started with boards me and my friends own, and all had gd32 chips. But I think a stm32f103rct6 or a gd32f103rct6 would work the same.

So I have the feeling the pin mapping is still not correct in your case?

vamfun, 2021-01-12:
Very observant, I didn’t notice the 6 wire halls.   Well, I’m not sure why you think my pins are not quite correct.   I have verified by tracing all but the charger pin  which alludes me still.    My chip has ST marking but still might be a  GD clone with ST markings.    

http://vamfun.wordpress.com
Robot musing blog
iPhone

closed 2022-01-01 by EFeru

dnhkng, 2023-09-28:
@vamfun Did you solve this issue?  I'm also using a 6-wire sensor hoverboard controller board, and plugging in the charger turns the system on, and powers the wheels!
`#define BOARD_VARIANT           1 `

If I then turn the system off using the power button, the motors shut down, but the buzzer runs constantly! No way can I charge it like that!

 If I just leave the system on to charge, after a minute or two, there's a time-out, and the system tries to power down. But again, I get a constant loud beeping!

Candas1, 2023-09-28:
Because your board has a problem.
It's described on the troubleshooting section of the wiki.

dnhkng, 2023-09-28:
I made an edit to the code, to disable the beep. Do you think it's necessary to repair the board? The LEDs remain lit, so there is a current draw and the batteries will take longer to charge, but if that's the only issue, I can live with that for now.