EFeru/hoverboard-firmware-hack-FOC #129: Working board suddenly stops listening to St-link

vamfun, 2021-01-21:
I was doing fine with my HOVER-1 H1 board until I decided to try some basic IO programs made from scratch with STM32CUBE.  I was able to read the ADC and set up the uart3 to print values.  Then I decided to try testing DMA Adc to memory and memory to memory transfers.  Something happened and now I cannot communicate with the board.  I verified continuity between the st-link and STM32 pins and tried a few other St-link clones.  The board 3.3v is present.   
Are there any other things I can try before pronouncing it bricked?

Candas1, 2021-01-21:
So you are sure the chip is powered ? 
If for example you delete the memory from ST-Link Utility, there is no program anymore so the latch is not held.
In this case I have to keep the button pressed to keep the chip powered.

Have you tried that ?

vamfun, 2021-01-21:
@Candas1 yes I do hold the switch closed.  Also the presence of 3.3v was verified on the chip itself.

Btw did u see the strange board I mentioned...something new for your board spreadsheet.

Candas1, 2021-01-23:
Is there a change you used the programming pins in your code ?

vamfun, 2021-01-24:
Certainly not on purpose😎. All I did was select the analog in and uart 3 pins on the Stm32cube configuration section.  The first program ran a straight adc read and print out over uart channel.  It ran fine .  I then modified the program to do a DMA adc to memory and a DMA memory to uart 3 transfer with a large block of text printing from memory.  I don't believe that DMA program ever downloaded because the ST-link stopped recognized the board.  
So no I did not use any programming pins unless that was generated somehow by the STM32CUBE software during compile.

I tried erasing with Keil and the St-link tools and using several different st-link clones.  No luck.

But let's say a programming pin write happened.  Can the chip still be reset or erased somehow?

Candas1, 2021-01-24:
I think this video saved me some time ago, maybe you can try doing the same:
https://www.youtube.com/watch?v=jEz0C2bT2M0

EmanuelFeru, 2021-01-24:
Nice video! I didn't know that.

vamfun, 2021-01-25:
Ok @Candas1  I officially love you for that link!  I'm back in business with a working st-link.
 
Turns out STM32CUBEMX is the culprit....After looking into it , any pins that are not specified in your configuration input are disabled by default...this includes SWDIO pin!!  So the easy fix is to set the sys configuration to any option but "no debug". If you specify a debug type the programming pins are activated.   I proved that this fix works.

It took me a while to get the timing just right on the reset button or grounding of the NRST pin... I have to start the erase and within 1 sec release the reset ground or the erase fails.   
On my chip pin 7 is the NRST pin and it has a pull up resistor connected to 3.3v and a capacitor connected to ground.  I grounded the resistor/capacitor junction rather than the chip pin.  
Again thanks for the link😎

Candas1, 2021-01-25:
I am happy that I could help.
I found out about that while investigating how to use SWDIO and SWCLK pins for something else, without preventing a reflash.
It seems you can do that if you allow some time at startup for flashing. And this trick helps if things go wrong.

No worries it's good to have spare boards for your projects

closed 2021-05-12 by Candas1