EFeru/hoverboard-firmware-hack-FOC #130: Piggyback motherboard variant in IO Hawk hoverboard

vamfun, 2021-01-21:
Since my hoverboard Hover-1 H1 motherboard quit (#129) I got a $35 used IO Hawk to hack.  Unfortunately it has the STM32 mounted on a separate board that plugs into a motor driver board like a shield. [image]
Bottom of stm board showing sockets.
[image]
Here is the driver board
[image]

This is a 4 to 5 year old board and it uses Infineon ir2136s drivers.   

I'm reluctant to flash it since the configuration appears quite different.   
Has anyone had experience with this ?

Teardown blog post [](https://vamfun.wordpress.com/2021/01/20/flying-firehose-blog-2-io-hawk-hoverboard-teardown/)

Candas1, 2021-01-21:
Hi, 

It's a monster, I have never seen such a mainboard before.

Even the chip is different, it has 100 pins it seems.
[image]

I am not an expert, but I imagine it would take a lot of time to make it work and you might not encounter such a board again.

As you said I would restore it to a hoverboard.

EmanuelFeru, 2021-01-24:
The `STM32F103VCT6`(your board) and  `STM32F103RCT6`(supported one), they both make part from the same family `STM32F103xC`. So, the properties seem to be similar, but as Candas said I doubt that it will work out of the box.. and I don't recommend trying it to avoid damaging something.

Overall, the board looks as better quality and I think upon some checking of the pinout, and configuration it should not be very difficult to port the code to this new board... it is just a matter of who is willing to take the challenge 😊

closed 2021-07-24 by EmanuelFeru

Candas1, 2023-06-02:
I ended up with like 20 mainboards with stm32f103vct6 chips directly on the same pcb.
Looks like I am the one to take the challenge.
Stay tuned

Any2suited66, 2025-02-26:
I know it's been awhile but were you able to flash the firmware on this board, @Candas1.  I've got two of them in my garage.