EFeru/hoverboard-firmware-hack-FOC #131: VARIANT_USART: util:calcInputCmd() sets speed to 0 :-(

RoboDurden, 2021-01-24:
Hallelujah. took me nearly two days and over 30 compilations to find out that the calcInputCmd was setting my speed to 0.
First issue is that you define the typ to be 3 in https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/63276132a7f97457a2bd19a29451e88569920985/Inc/config.h#L313
But there is no case 3: in calcInputCmd()

And it seems that you store input1 and input2 in the eeprom and then the #define PRI_INPUT2 in config.h is ignored anyway ?

I guess i crashed my eeprom the last time and since then i was no longer able to send speed commands to the board.
For now i have hard coded the #define:

```
void calcInputCmd(InputStruct *in, int16_t out_min, int16_t out_max) {
  in->typ = 2;
  in->mid = 0;
  in->dband = 0;
  in->max = 1000;
  in->min = -1000;
```

**Would be nice if someone can tell me how to restore everything to normal.**

P.S. it might be good to have a global error variable and a beeping of various error codes.
To simply end the switch clause with 
```
    default: // Input is ignored
      in->cmd = 0;
```
is aweful. If you had such a global error variable could at least set it to some new integer whenever you quickly add code :-)
Then every user could post the error code here and you would quickly find the source.

P.P.S i have written a very nice IDE for my online compiler: https://pionierland.de/hoverhack/
As i use git commands to clone the repos, i might be possible that now everyone could issue pull request directly from my online compiler.

juodumas, 2021-01-24:
> To simply end the switch clause with is aweful.

@RoboDurden please try to be more respectful and use nicer language.

RoboDurden, 2021-01-24:
**But it is aweful.**
There should be a `.case `for every `.typ `and the `default:` only for a wrong system config. And that should output an error and not simply set the input to zero.

**Sorry, i can't stand hypocrisy.**

Candas1, 2021-01-24:
Type 3 means that the type will be automatically detected.
At start-up, either you did auto-calibration and values are loaded from EEPROM, or if type is 3 it will derive the real type from the limits in config.h.
[Here](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/63276132a7f97457a2bd19a29451e88569920985/Src/util.c#L316) is where it happens.

So type 3 should actually never happen in the function your mentioned.
I am curious why you have type 3 in your case, it could be a bug, I am only human....
The firmware is highly complex and although I try each variant myself after a propose a PR, I can miss bugs.

Please let me know if you find why type is 3 there although it should be overwritten, I can check how to fix it.

But please be more respectful, you also make your software available as open source to others so you should understand. Treat people how you would like to be treated.

We are definitely open to feedback for improvements.

RoboDurden, 2021-01-24:
I only know that my typ is neither 1 nor 2 so `default:` is executed which sets input to 0 :-(
When i only set `in->typ = 2;` , the motors were spinning again but speed was wrong.
After i also set `->max` , `->min` , etc. everything was back to normal.
So i think i have some random numbers.
My debugging options are limited to the `Feedback` structure..

As it happenend out of nowhere after flashing a new firmware i can only guess that the previous flash failed and corrupted the eeprom.
Don't know why you put `TYPE, MIN, MID, MAX, DEADBAND `into the eprom but i guess some `VARIANT_XY `allows calibration of `MIN, MID `and `MAX.`
Doesn't really make sense for `VARIANT_USART.`
I will try changing  `#define FLASH_WRITE_KEY      0x1102 ` tomorrow, maybe this will reset the eeprom.

I only criticize when i can offer a solution.
So stop whining and be happy to change the `default:` behavior :-)
It took me a lot of time (time = happiness) to trace that bug.

P.S. if i had a bible, it would be **Fight Club**.

Candas1, 2021-01-24:
I wanted to make auto-calibration possible for all variants, so you could still benefit from this feature with usart variant if you have your inputs on an ESP32 for example and you want to calibrate your throttle handle or joystick.

Yes changing the FLASH_WRITE_KEY is another solution.

RoboDurden, 2021-01-24:
I will copy the InputStruct values into the Feedback struct tomorrow and we will see if they are random numbers.
Good night from a snowy Germany :-)

RoboDurden, 2021-01-24:
>  benefit from this feature with usart variant if you have your inputs on an ESP32 for example and you want to calibrate your throttle handle or joystick.

I don't think it is a good idea to simply forward the adc values from a joystick to the hoverboard controller and calibrate the min max values there. It is way more clean and straight forward to scale them to +-1000 on the esp32. And the esp32 code might want to know when the joystick is at max too.
I also remember another issue here were users did not understand why they have to calibrate when using VARIANT_USART.
It is counter counterintuitive, (With an oop mindset, everything should be intuitive..)

And yes sorry for being rude again, this is again the typical stupid Linux mindset were the goal is to allow all thinkbale paths.
The linux believers still live in the dark ages before oop.

But of course i might have misunderstood your ESP32 example :-)

Candas1, 2021-01-24:
I am not saying you have to use auto calibration with usart. You can use it if you want to, and code is the same for all the variants.

Usart works without the calibration. I am using it myself.

RoboDurden, 2021-01-25:
Another day in toy world.

> I am not saying you have to .. You can use it if you want to

That is the exact wrong mindset that i criticize.
The tragedy of the toy world ruled by hypocracy is that no-one really listens to the other:
> > And yes sorry for being rude again, this is again the typical stupid Linux mindset were the goal is to allow all thinkbale paths.

So i added to main.c

```
extern UART_HandleTypeDef huart2;
extern UART_HandleTypeDef huart3;
...
            Feedback.speedR_meas	= (int16_t)input2[inIdx].mid;
            Feedback.speedL_meas	= (int16_t)input2[inIdx].dband;
            Feedback.batVoltage	    = (int16_t)input2[inIdx].min;
            Feedback.boardTemp	    = (int16_t)input2[inIdx].max;
            Feedback.cmdLed     	= (uint16_t) input2[inIdx].typ,
``` 

and got

typ:    0
min:    -300
mid:    -115
max:    -115
dband:  0

after resetting  `#define FLASH_WRITE_KEY      0x1003` it returned to 

typ:    2
min:    -1000
mid:    0
max:    1000
dband:  0

and now everything is back to normal.

Unfortunately -300 and -115 do not look like random numbers from a corrupted eeprom. So there must be a real bug.

And a final word on what a bug is.
I remember an automation company that build very expensive robot automation for the top car companies in Germany.
And they nearly lost all their good reputation because a programmer added a check function that did only allow parts to be put into the machine when the two barcodes did match. However he did not take into account that when a part had no barcode, the code was read to 0 and with 0 == 0 = okay a total destruction followed.

Now even so this is a bug, it is human to overlook it. 
But adding 

    default: // Input is ignored
      in->cmd = 0;
      break;

did not happen by accident. It was deliberately added to take care for something that should not happen at all.
And that is really bad programming style. Not to call it aweful again :-)
And as this code is running on the open streets, my firsthand story fits.
Of course you now will object that setting speed to 0 makes everything safe.
But firstly i had already this problem with your firmware make full breaks when i feared them most.
And secondly is the actual bug not the true subject but how you deal with it:

> PS: I didn't read all your comment because it's too long and not to the subject. Cheers.

Roland,
the born loser

"you either lose.. or fail." and "losing liberates" (Robo Durden)

"It's only after we've lost everything that we're free to do anything." (Tyler Durden)

RoboDurden, 2021-01-25:
>You just entered auto-calibration by mistake. Maybe while holding the button when flashing.

Yes i sometimes forget to short the two button pins and the flash gets stuck in an endless loop that i interrupt by pulling the usb dongle. But that is something you have to expect when offering your code to everyone.

> So it has nothing to do with your rant about having a default case... Here it works as expected, the problem is somewhere else.

No the problem is that i halted my project for 2 weeks because i had lost all the happiness when out of nowhere the board did no longer respond to the serial commands. And you can count me as an experienced user.

If a `case 0:` indicates an unwanted auto-calibration then the action must be an `error code` and not `in->cmd = 0;`
You are not stupid, you know it is a bug. The problem is that you are not willing to admit a mistake.
Especially to someone who (intenionally) behaves like an ass-hole.

"you either lose.. or fail. nothing in between, nothing else."

But fear rules the world.
And everyone fears his weakness. 
So everyone has to be equal and hipocracy is everything :-(

Candas1, 2021-01-25:
Everyone is using the firmware in a different way.
They are ebikes, scooters, robots, mowers, gokarts, ball throwers for dogs.....
Some people only use a single input so they can ignore one of the inputs.

So there can be problems and we are here to help. You don't have to struggle for 2 weeks and come back frustrated as you are.

RoboDurden, 2021-01-25:
yeah yeah, that is again exactly the problem:

> So there can be problems and we are here to help. You don't have to struggle for 2 weeks and come back frustrated as you are.

It is this typical mindest of these Linux believers to choose a well enclose field, become the only expert and then 'generously' help their followers.
With the mindset of 21st cenutry programming = oop, everything has to be so intuitive that followers can solve problems themselves.

I indeed have noticed, that you are very helpful here. But i do not come here for help because i do neither deserve it nor want to eat up your prescious time.
I only come here when i have found a problem and can offer at least a partial solution.

Problem here is your bad programming style.
Partial solution offered is a global error code variable that gets beeped out.

You even have already the beeping in https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/63276132a7f97457a2bd19a29451e88569920985/Src/main.c#L472

BLDC_controller.h probably would be the best place for  `uint8_T iErrorBug = 0;`

P.S. i am not frustrated. in my world it is only about solving problems. So the more direct a problem is depicted the better.
And the more all you toy people stick to your hipocrasy, the more ass-hole is needed to test you again and again.

I lose.. you fail.

Candas1, 2021-01-25:
This is not my repository. I am just a user like you that got to use the firmware and proposed some enhancements.

RoboDurden, 2021-01-25:
That's bad. I was happy to imagine that Emanual had found/allowed some collaborators.

Candas1, 2021-01-25:
Anyone is free to propose Pull Requests, this is how that firmware was built.

RoboDurden, 2021-01-25:
So i am not like you. I have never proposed a pull request (and will not do so) because i miss the willingness to improve the code.
So i always first start an issue like this to give the toy world another chance..

The problem with the stupid is that they are too stupid to at least accept the possibility that they are too stupid to see that they are too stupid - which can be seen as the most basic definition of stupidity.

Of course you think that i am the stupid guy. But you ignore that if i wouldn't be it, you would not be able to see it anyway.
The tragedy is that your ignorance is stupidly based on the hipocracy feigning greatness in each other.

you either lose.. or fail :-(

Candas1, 2021-01-25:
Sorry this discussion is not about the issue anymore.
I will check if I can reproduce it and propose a fix as this might affect other users also.

RoboDurden, 2021-01-25:
> Sorry this discussion is not about the issue anymore.

It was no longer about the bug for already 2 of your posts. So why now sorry ?
But yes, we can close this issue :-)

closed 2021-01-25 by RoboDurden