EFeru/hoverboard-firmware-hack-FOC #134: Inhibit Autocalibration capability

vamfun, 2021-01-27:
From @RoboDurden issue #133
"And the good old cmd1 and cmd2 are turned into
InputStruct input1[INPUTS_NR] = { {0, 0, 0, PRI_INPUT1}, {0, 0, 0, AUX_INPUT1} };
InputStruct input2[INPUTS_NR] = { {0, 0, 0, PRI_INPUT2}, {0, 0, 0, AUX_INPUT2} };"

I must confess @RoboDurden has a good point here... I ran into similar debugging headaches because of this complexity.   As a controls engineer I like the cmd1 and cmd2 readily visible in Main.c.

I too inadvertently had a zero output from the auto calibration.   I thought I was ok because I selected type 2 ADC since I am using the ADC to drive the motors as servos and also don't want auto cal.   

But I may be mistaken, seems an auto cal is always performed if the power pin is left in for 5 seconds even if Type 2 is chosen.    

Perhaps if you make a future mod in this area I would recommend giving the user the option to disable auto cal with a define.   

The reasoning for me at least is to prevent ignored inputs if during debug you simply left the power jumper in for a few seconds too long and didn't move the input.  A zero command results and is committed to memory unless a full erase is done.   I am working with a flying object and need to have very reliable inputs during testing and I know I might be distracted and jumper the power pin enough to enter auto cal.

Candas1, 2021-01-27:
Hi,

Yeah this was my idea...
Before it would have also entered auto calibration but would have failed if inputs were not moved.
I myself tested it many times without issues but I am using a push button... 

Now that I know that some people use jumpers it's definitely a problem. 
We were thinking about preventing entering auto calibration at startup so that it's not happening after having flashed the board. It would enter this mode only if button is pressed later. Do you think it would work in your case?

Candas1, 2021-01-27:
Do you mind trying adding this line of [code](https://github.com/Candas1/hoverboard-firmware-hack-FOC/blob/d5d85c20ec76a5d2273e623f759d6b884410cce7/Src/main.c#L203) in main.c ? 

When the board is ON, if the button is pressed or you put a jumper for flashing, it will not enter auto-calibration as the button won't be released. (n change here )
But after the board is flashed and restarts, it will wait until the button is released. It worked for me now but would you be so kind to help testing this change in case I overlook something ?

vamfun, 2021-01-28:
Thanks for the input....but for me it is not a strong enough inhibit
I added your suggestion but also modified my code to inhibit any power  button calibrations or adjustments unless specifically enabled in a define.     Any adjustments to the adc calibrations can cause major unexpected  input transients...so for safety added these extra constraint.
Now if the power jumper is replaced after the board is running... it simply stops until the jumper is removed if no auto cal is enabled.  Then the board simply continues to run.   

Here is what i did

config.h  added these defines to default section

//Auto calibration defaults are off ... change if you want to auto calibrate your ADC inputs or adjust current and speed
#define ADC_AUTO_CAL_ON                0    // if you want to auto calibrate ADC inputs set "ADC_AUTO_CAL_ON  1"
#define MAX_CURRENT_SPEED_ADJUST_ON    0    // if you want to adjust maximum current and speed on with button "MAX_CURRENT_SPEED_ADJUST_ON 1"
// ######################### END OF DEFAULT SETTINGS ##########################

******** IN poweroffPressCheck  i added to your if tests ***********
void poweroffPressCheck(void) {
  #if !defined(VARIANT_HOVERBOARD) && !defined(VARIANT_TRANSPOTTER)
    if(HAL_GPIO_ReadPin(BUTTON_PORT, BUTTON_PIN)) {
      enable = 0;
      uint16_t cnt_press = 0;
		
      while(HAL_GPIO_ReadPin(BUTTON_PORT, BUTTON_PIN)) {
        HAL_Delay(10);
        if (cnt_press++ == 5 * 100 ) { beepShort(5); }
      }
      if (cnt_press >= 5 * 100   ) {                         // Check if press is more than 5 sec
        HAL_Delay(1000);
******************************************************************************************************************
        if (HAL_GPIO_ReadPin(BUTTON_PORT, BUTTON_PIN) && MAX_CURRENT_SPEED_ADJUST_ON) {  // 
*******************************************************************************************************************
Double press: Adjust Max Current, Max Speed
          while(HAL_GPIO_ReadPin(BUTTON_PORT, BUTTON_PIN)) { HAL_Delay(10); }
          beepLong(8);
          updateCurSpdLim();
          beepShort(5);
*******************************************************************************************************
        } else if (ADC_AUTO_CAL_ON) {               // Log press: Calibrate ADC Limits if enabled
*********************************************************************************************************
          beepLong(16); 
          adcCalibLim();
          beepShort(5);
        }
      } else if (cnt_press > 8) {                         // Short press: power off (80 ms debounce)
        poweroff();
      }
    }

See what you think

commit a564f368 (2021-01-29): Ascii Debug updates

- added missing pointer to UART2 input user processing
- added comms.c to Makefile
- fixed #134 : uncomment #define AUTO_CALIBRATION_ENA to disable auto-calibration
- small styling adjustments
closed 2021-01-29 by EmanuelFeru