EFeru/hoverboard-firmware-hack-FOC #136: ADC variant does not work...troubleshooting?

epikao, 2021-01-29:
Hello,

I ordered two hover electronics boards as shown on the Emanuel front page via Aliexpress + a Nunchuck controller.

With the Nunchuck controller, reverse and forward does not work cleanly. If I push the joystick backwards or forwards, the motor (both) turns mostly in the right direction but not always, and that only one motor turns somehow does not work with the joystick.

I have now programmed the board with the ADC variant, but unfortunately here nothing works.
Interestingly, I have both variants no sound from the buzzer, is that normal?

I just do not know where to look for the error? The board can be switched on, at least an LED lights up.

Or are these Aliexpress boards defective?

Would appreciate any hints.

Many thanks

Candas1, 2021-01-29:
Do you have a picture or link of the board just in case ?
Not having sound is weird... Could be that the board is little different.

There are little instructions in the [wiki](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/Variant-ADC)

epikao, 2021-01-29:
thanks for your reply. See attached pictures. 
[image]
[image]
[image]

I'm meaning the below board

epikao, 2021-01-29:
and here the l

epikao, 2021-01-29:
I uploaded the generated firmware.bin file via ST-Link software. The build I done with vscode/plio

EmanuelFeru, 2021-01-29:
@epikao Seems like you have a slightly different board  version, at least at this middle location:
yours:
[image]
vs. mine:
[image]

The coil is a different location. And I am sure because you don't hear the beep and melody, some pins are also changed.
With a multimeter can you identify if your pins are corresponding to [defines.h](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/master/Inc/defines.h)?

If not, then you need to do a small exercise to update the pins similar to [here](https://github.com/NiklasFauth/hoverboard-firmware-hack/issues/162) (but with your pins of course).

Candas1, 2021-01-29:
OK I was too slow and Emanuel replied ah ah
Yes we expect PA4, on some boards it's mapped to PC13, but those boards had other differences as well.

You can also program directly from vscode/platformio.
To debug ADC you can use [serial debug](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/Debug-Serial)

EmanuelFeru, 2021-01-29:
Luckily, I was around this time :P

epikao, 2021-01-29:
many thanks, I'm checking all that now. But some more questions:

Does the controller send always some data over the serial? Till now I didn't had success to receive anything.. I'm also not sure if I need the battery power for that, or if only supply over my programmer is enough. With supply of programmer and no battery power the LED switches directly on, so the on/off switch has no function. 
The buzzer is connected via transistor Q6 and Resistor R56 to PA4, so this connection should be according default code, but still no sound. I think the buzzer only works with battery supply because of VSW = 5V? Anyway also with battery supply no sound. Looks I must measure the signal output of PA4....

epikao, 2021-01-29:
oh Buzzer looks is connected at PA6 not PA4, is that a problem?

Candas1, 2021-01-29:
Motors are not spinning for safety reasons at startup if cmdL and cmdR are not within -50 to +50 range.
You have to do [input calibration](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/Input-Calibration) .

epikao, 2021-01-29:
> Motors are not spinning for safety reasons at startup if cmdL and cmdR are not within -50 to +50 range.
> You have to do [input calibration](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/Input-Calibration) .

ok, I will test this later, now I first want make work this buzzer.
I took my second board, here I found out that with original firmware, the buzzer also works only with programmer supply. That is strange, because according schematic the buzzer should be connected to VSW, and this is also the case on my board. 
After uploading with the Firmware from this side, no more sound... hmm maybe something with the frequency.

Candas1, 2021-01-29:
Hmm weird, is it a problem on the first board.

epikao, 2021-01-29:
so yes, something at the buzzer is defective on my first board. 

With the second board buzzer works, and motor are spinning, but both motor spins always together. I couldn't find a position where the one motor stops and the other one is spinning... why that?

And I also not understand that even buzzer is connected at PA6 it works if Software is set at default PA4...

Candas1, 2021-01-29:
It's not a "tank" style mixing.

[image]

epikao, 2021-01-29:
ok, but I also couldn't find a position where both motor are spinning in the same direction, always spinning in opposite... but will check tomorrow again more carefully. Thanks for your great support.

EmanuelFeru, 2021-01-30:
Making the buzzer work is a confirmation that the interrupt for motor control is working correctly. So it's a good idea to fix the issue.

Are you sure the buzzer pin is correct? And what chip do you have?
Check buzzer for continuity? Did the board work with factory firmware before you flashed the chip? I always do one check with new boards just to avoid surprises.

epikao, 2021-01-30:
> Making the buzzer work is a confirmation that the interrupt for motor control is working correctly. So it's a good idea to fix the issue.
>
> Are you sure the buzzer pin is correct? And what chip do [...]

Thanks! First board with original firmware I can not remember that there was sound from the buzzer, but with my second board. So the buzzer electronics on the first board must have a defect somewhere. The confusion regarding PA4, PA6, was because I checked the wrong controller datasheet.

Unfortunately, I still don't fully understand the spinning behavior of the motors in combination with the states of the ADC inputs:

Following my setup and findings. I never had a state where both motors are spinning in the same direction... also I not understand why some input states make motor not spinning very smoth or very slow, or beep, and sometimes stopping...

[image]
[image]
[image]

vamfun, 2021-01-30:
That is a mystery... Best thing to do is uart print cmd1 and cmd2 to see if they agree with your motor movement.   I think your setup will introduce a lot of noise into the ADC which may account for some of the roughness.   Try grounding all inputs as close to the input plug as possible then run a debug with that input.   Also what steer/speed blending are you using?
How did you specify the type of input...1,2 or 3?  Did you enter autocal?
Maybe use the new repository , comment out auto cal enable , then start with a full erase to make sure auto cal values are not affecting your inputs.  Set your type as 2 and see what you get.

epikao, 2021-01-30:
#Also what steer/speed blending are you using?
How did you specify the type of input...1,2 or 3? Did you enter autocal?

I did the autocal, but not more, I also not study the other setups, just download the newest code from this respository, build ADC variant and upload... so all other setups should be still default. I was hopping that default should work, but it looks I must study now your mentioned other parameter, and I will record the UART output and maybe do again an autocal...

epikao, 2021-01-30:
`Also what steer/speed blending are you using?`

And what do you mean with that? I'm using 0V, 3.3V and two 33K resistors to each input do genarate 1.65 V (MID), see setup above.

vamfun, 2021-01-30:
As I understand it, the default blend is x.5 on steering (blue adc1) and x1 on speed (green adc2) so if green is 1 or -1 the blue cannot cause a reversal until green is at mid. But this doesn't explain your table.   The shutdowns were probably caused by emergency shutdown due to exceeding the ADC timeout count (indicated by beeps) due to high level of noise.
I would also guess that you may need to reverse a hall sensor lead (or change direction in sw) to get motors running in same direction.

EmanuelFeru, 2021-01-30:
@epikao, for testing purposes I think is easier to use 2 potentiometers and remove the 33k resistors. Because auto-calibration requires that you move the potis thorough the whole range.

And if your motors act roughly it could be that 2 of the motors phases are swapped. Sometimes the colors don't match. Swap the phases and give a try with low input.

Overall to get the ADC variant  running it should be:
- prepare your hardware, connect potis, etc.
- full chip erase
- flash ADC variant
- run auto calibration
- the motors should run

epikao, 2021-02-01:
`I would also guess that you may need to reverse a hall sensor lead (or change direction in sw) to get motors running in same direction.`

besides EMC, I think this is the main problem.
I have currently already tried different motor phase connection combinations - unfortunately still without success.

Do I have to try the Hall connections as well? Can I not either or? So Hall or motor phases ? It is easier to change the motor phases. Maybe I should also check if the Hall-Sensors are not connected on other pins... but I think then the motor would not spins...

What is the easiest way to find the right combination? Try and error? I could also open the motor cover....

EmanuelFeru, 2021-02-02:
I think try and error is the easiest. The other way is to use an oscilloscope and log back-emf voltage with Hall sensors. The should match the [motor mapping](https://raw.githubusercontent.com/EmanuelFeru/bldc-motor-control-FOC/master/02_Figures/control_methods.png).
However, from what I heard until now just inter-changing motor phases to find the right combination was sufficient. I didn't hear yet that Hall sensors need to be inter-changed... but I don't exclude it.

epikao, 2021-02-05:
ok thanks, I couldn't find a combination that works...(I tested with the Nunchuck Variant, with 15V to 3.3V regulator between). At some combination the motor started without touching the Nunchuck, often with very strong shaking. Crazy. Could it be that the mosfetdriver maybe are some on the wrong GPIO's ? I fear that I must now check the whole PIN assignment...

epikao, 2021-02-07:
I think, I've found the problem. The hall wires Left, are connected to motor phase Left Hi, and so one, so something is swapped and not according pin assignment....

Candas1, 2021-02-15:
Can you record a video of how it behaves ?

closed 2021-07-24 by EmanuelFeru