EFeru/hoverboard-firmware-hack-FOC #138: Improved Main_Loop_Delay control

vamfun, 2021-02-08:
Problem...there are now rate limits and filters based upon the periodic loop delay.   This is currently done by a pure HAL_delay(main_loop_delay) function.   The total main loop delay is actually around 1 ms extra due to the delays from execution of the code the loop.     I determined the 1 ms based on timing measurements with my ADC variant modified PID code .     As a controls engineed,  i want precise control of timing loops so filters and rate limits can produce predictable results.     I first made a  micro second timer function and used it to add the exact delay needed over and above the 1ms  make the  main_loop_delay (ie. 5 ms).     I had to use the  debug clock  to create a  delay_periodic_loop( main_loop_delay) function that replaced the HAL_delay() in the main loop.   This was very simple but required use of another timer.   I then had the idea to make use of the bldc 1600 hz interrupt  based buzzerTimer variable.     This timer has 16 ticks per ms so i added an If test in the main loop that will only allow execution if buzzerTimer > 16*main_loop_delay.

This works great for the 5ms main loop delay since it frees up 4 ms to be used for printf debug.   My testing shows that every 8 printf characters adds 1 ms of main loop delay , so I can now pring 32 characters on every main loop step and still maintain and exact 5 ms total loop time.      
My reommendation is that this simple change would improve main loop timing.

Here are the  two code approaches 
**First Method uses debug timer to creat delay_periodic_loop**
in main.c
... 
//HAL_Delay(main_loop_delay);   // replace this with
delay_periodic_loop(main_loop_delay)
...

in utility.c  
....
//void delay_periodic_loop( volatile uint32_t loop_ms)
//{
//  //This routine creates a ms delay for a periodic loop.   It uses a micro second debug clock in a while loop to wait until the total
//	//loop time equals loop_ms.  It differs from a straight HAL_Delay which simply adds a fixed delay to other loop delay.
//	//This routine allows the delay to be smaller to compensate for larger loop delays caused by printf etc.
//	static uint32_t initial_ticks = 0;
//	uint32_t ticks = (HAL_RCC_GetHCLKFreq() /1000000);
//	uint32_t loop_ticks = loop_ms*1000*ticks;
//  static _Bool bflag = true; //initial pass true
//	if(bflag)
//	{
//		CoreDebug->DEMCR &= ~CoreDebug_DEMCR_TRCENA_Msk;
//		CoreDebug->DEMCR |=  CoreDebug_DEMCR_TRCENA_Msk;
//		DWT->CTRL &=  ~DWT_CTRL_CYCCNTENA_Msk;
//		DWT->CTRL |=   DWT_CTRL_CYCCNTENA_Msk;
//		
//		printf("initial_ticks %i loop_ticks %i ticks %i ",initial_ticks, loop_ticks,ticks);
//	  bflag = false;
//		DWT->CYCCNT = 0;
//		//HAL_Delay(1);
//		//printf("CYCCNT after 1 ms %i ", DWT->CYCCNT);
//	}
//	//initial_ticks = initial_ticks + loop_ticks;
//		while ((DWT->CYCCNT - initial_ticks ) < loop_ticks); 
//	  initial_ticks = DWT->CYCCNT ;
//	//printf("initial_ticks %i CYCCNT %i ",initial_ticks, DWT->CYCCNT);
//	
//	
//}

**Second Method uses buzzerTimer variable** 
in main.c 
while(1)
if(update_periodic_loop) { execute main code ...) 
else { go to end while)

in utility.c
_Bool update_periodic_loop( volatile uint32_t loop_ms)
{
  //This routine creates a periodic loop based upon elapsed buzzerTimer ticks.    buzzerTimer is updated in BLDC.c and has 16 ticks per ms.   buzzerTimer is now a global and used in main.c to make the  
//main loop time equals loop_ms.  It differs from a straight HAL_Delay which simply adds a fixed delay to //other loop delay.
//This routine allows the delay to be smaller to compensate for larger loop delays caused by printf etc.
	
	
	
	static uint32_t loop_ticks ;
	static uint32_t last_buzzerTimer = 0;
  static _Bool bflag = true; //initial pass true
	if(bflag)
	{
		loop_ticks = loop_ms*16;
		last_buzzerTimer = buzzerTimer;
		bflag = false;
	}
	// Return true if at least loop_ms have elapsed since last loop was run.
		if((buzzerTimer-last_buzzerTimer )< loop_ticks){return false; }//{printf("%i ",buzzerTimer-initial_ticks);} 
	  else
		{ 
			last_buzzerTimer = buzzerTimer ;
			return true;
		}
	
	
	
}

EmanuelFeru, 2021-02-08:
@vamfun very cool idea! I happy you brought it up. I was thinking for a long time how to get rid of the blocking delay in main loop and you found 2 solutions. I like better the one with `buzzerTimer`.
Some functions use `main_loop_counter` so it would be ideal to keep this counter. The way I see the implementation in very easy terms. Just remove this:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/36953c6fe9ba90593d21150113459b0a4bb4e312/Src/main.c#L206
and wrap the main loop into
```c
if (buzzerTimer % 16*DELAY_IN_MAIN_LOOP == 0) {   // 1 ms = 16 ticks buzzerTimer
    // execute main code
}
```
Do you see any issue with this approach?

vamfun, 2021-02-08:
Yes ,  I believe your approach has some problems when the printf causes a loop delay of  more than 5ms.   What will happen is there will be additional waits until the next frame is allowed .     Lets say that there is a 6ms loop time, My approach allows the loop to run as fast as possible at 6 ms per loop.       After the first 6ms frame, yours would wait until 10ms to start the next frame, and the following frame would start at 20  ms.      Yours will complete 3 frames in 26 ms, mine would complete in 18 ms.

EmanuelFeru, 2021-02-08:
Makes sense. Then if we update to:
```c
if (buzzerTimer - buzzerTimer_prev > 16*DELAY_IN_MAIN_LOOP) {   // 1 ms = 16 ticks buzzerTimer
    buzzerTimer_prev = buzzerTimer;
    // execute main code
}
```
it should be ok. I am sure your code does similar. I am just thinking to get the same functionality in a minimum lines of code.

vamfun, 2021-02-09:
Yes, this is basically my code.  However , I chose to hide the buzzerTimer logic from the main code with a function to simplify the readability.   This code is going to be used by all types and it is not necessary for them to understand why a buzzer?? related timer is used in main loop timing.  They don't need to know why the 16 is in the logic, and if they want to fully understand that code they are obliged to dig into BLDC like I had to do.  You are familiar with the code and it makes perfect sense with a quick glance but probably not to others.    

If(update_period_loop) // creates a ms period = main_loop_delay
{ }
 
I'm not a coder ...so this kind of statement is easier to understand at the cost of a few extra lines of code to create a function.

Of course, this is a matter of taste and your code will work fine...so no big deal😎.  I appreciate the way you collaborate and help others , in particular me!

EmanuelFeru, 2021-02-09:
I also appreciate proposals, suggestions for code improvements. I am sure community appreciated it too :) So, thank you!
I will implement the change in the next commit ;)

vamfun, 2021-02-09:
Regarding timing delay overruns... probably wise to split up the printf  statement to keep loop timing under 5ms.   Not sure what other code variants do , but as I mentioned I get a 1 ms added delay for every 8 char printed.    So in my code ... I keep prints less than 32 char.   Here is what I did
``      if (main_loop_counter % 25 == 0) {    // Send data periodically every 125 ms
        printf("in1:%i in2:%i cmdL:%i cmdR:%i \r\n",		
          input1[inIdx].raw,        // 1: INPUT1
          input2[inIdx].raw,        // 2: INPUT2
          cmdL,                     // 3: output command: [-1000, 1000]
				  cmdR );                   // 4: output command: [-1000, 1000]
				}
				if (main_loop_counter % 260 == 0) {    // Send data periodically every 1.3 ms	
				printf("BADC:%i BV:%i TADC:%i T:%i\r\n",
          adc_buffer.batt1,         // 5: for battery voltage calibration
          batVoltage * BAT_CALIB_REAL_VOLTAGE / BAT_CALIB_ADC, // 6: for verifying battery voltage calibration
          board_temp_adcFilt,       // 7: for board temperature calibration
          board_temp_deg_c);        // 8: for verifying board temperature calibration
				}

closed 2021-07-24 by EmanuelFeru