EFeru/hoverboard-firmware-hack-FOC #139: Link to Hoverboard-Hack-FOC-PID-Servo

vamfun, 2021-02-10:
Not really an issue but @Candas1  mentioned that this variant might be worth evaluating.
[https://github.com/vamfun/Hoverboard-Hack-FOC-PID-Servo]

I modified a fork to  allow ADC  inputs to  PID control the motors as 360 deg servos using the Hall sensor feedbacks.    Unfortunately I am using a motherboard variant that has 5 GPIO changes but restore these to your board and you can try it.    Appreciate any feedback.

EmanuelFeru, 2021-02-11:
@vamfun the graph looks very cool! Do you have a movie to see live how is the motor behaving? I would like to see also smaller position targets.

Candas1, 2021-02-11:
Hi, I tried it yesterday but I forgot to record...
I had some problems with the firmware not reacting sometimes but I need to fix my ADC set up.

I will try again this evening.
It's really interesting for robotics.

Candas1, 2021-02-11:
Position control
https://youtu.be/VpphzLXatIo

Under load
https://youtu.be/Ai6PMY9mlt0

EmanuelFeru, 2021-02-11:
I like it too! 👍

Candas1, 2021-02-11:
And this was for fun lol
https://youtu.be/pM6i7bIdHPE

vamfun, 2021-02-12:
Glad to see it worked for you.  Wheel driving wheel is a form of power steering😎

I haven't got a good method of plotting yet..right now I'm using excel and it's a bit tedious.    I'll do a few smaller steps soon.

I've done some theoretical predictions based upon a first order motor response with tau = .02 calculated from motor constants and Wheel inertia.   Here is the block diagram for the PID control loop.   
[image]

Candas1, 2021-02-12:
And another one for fun:
https://youtu.be/QBDGHEpJk08

Inspired by this:
https://youtu.be/kCHiuNaOTdY

vamfun, 2021-02-13:
I add the smaller step responses to my original post.    

@Candas1 I found the links interesting.  The hoverboard is a perfect setup for your little servo slaving experiment.

pigiit, 2021-04-11:
Hi, I tried it for my board controlling the wheels with the RC and the receiver connected to the sideboard...I like it!
Great work!

vamfun, 2021-04-12:
Thanks...I’m adding these encoders for higher resolution.   They fit nicely on the interior of the wheel hub.  https://www.andymark.com/products/lamprey-absolute-encoder

http://vamfun.wordpress.com
Robot musing blog
iPhone

EmanuelFeru, 2021-04-12:
If you install an encoder, maybe you can try feeding the angle to motor control as well in bldc.c That functionality I did not test it yet, but it is there as input.

vamfun, 2021-04-13:
I saw where the input is used.   Are there any defines or switches that need to be set or do you comment out the hall sensors??

http://vamfun.wordpress.com
Robot musing blog
iPhone

EmanuelFeru, 2021-04-13:
There is no need because the hall sensors are ignored inside the controller itself.

gnbl, 2021-06-14:
Fantastic, just what I need: a cheap, high-torque, open source servodrive! So this now almost competes with VESC and ODrive.
Can this be integrated as "POSITION MODE"?
Power steering (allowing remote control) would be awesome with a steering torque sensor (two high resolution encoders and a torsion bar)...

gnbl, 2021-06-14:
> Under load
> https://youtu.be/Ai6PMY9mlt0

The noise is due to the low resolution of the position sensors?

vamfun, 2021-06-15:
> If you install an encoder, maybe you can try feeding the angle to motor control as well in bldc.c That functionality I did not test it yet, but it is there as input.
> 
> https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/0709e4cb62ef1d0bfedfc6bc3187614d5862acd6/Src/bldc.c#L188
> 
> 
> 
> And you need to activate the external angle measurement by making this parameter `1`:
> 
> https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/0709e4cb62ef1d0bfedfc6bc3187614d5862acd6/Src/util.c#L240

I'm getting ready to test the added encoders... I assume that the phasing of the encoder should match the electrical angle.

I'm having some issues with my motor rollover counting function.  I tried calling the function in main but the motor phase rate of change exceeds 180deg /5ms at maximum speed.   I tried moving the routine to the bldc.c 1600 HZ DMA1_Channel1_IRQHandler(void) { ... } but I'm sure its exceeding the .625 ms execution window .  Have you done accurate timing measurements of this loop to show what margin there is for added code?  I have a feeling margin is very slim.
I really just need to call it at 400 Hz.   I can modify main to accommodate this but it would be easier to just call it in a faster loop.

EmanuelFeru, 2021-06-15:
In DMA1 there is not so much margin, i think around 20% time from the 0.625ms calculation window. Especially Speed mode takes the most, followed by Torque, Voltage, Sinusoidal, and Communtation which is the lightest. There is an issue I think where I calculated the computational times with an oscilloscope, maybe you can find it.

vamfun, 2021-06-17:
Ok..rewrote rollover MotorPos routine and got the execution down from 54 us to 5 us with more efficient code so it now works in bldc.c.    Unfortunately I shorted my USB ports and blew the motherboard requiring a laptop replacement…twice !! within one week just before getting to the encoder testing plus the hoverboard went hard over and flew off my support box burning out a power supply and a $40 encoder.    Not a good week.   I going to have to limp along with my old xp laptop for awhile.  

http://vamfun.wordpress.com
Robot musing blog
iPhone

s56jsc, 2021-08-10:
Great job!
I try to upload it on my mower, but something not work.
I did modificate GPIO on defines.h for my board, but something not work property. When I power on in next second it turn off. Also when I build file I got some warnings
[image]

best regards
Simon

vamfun, 2021-08-11:
Hi Simon
Not sure why you are using the PID in your mower.   The command over usart is a wheel position command and it is scaled for 1000 counts equal 180 degrees.   -1000 to 1000 gives one rotation.       You can make it a rate command by setting Kp =1 and removing the wheel feedback.  Ie.  PID.feedback = 0. 

The shut off shortly after turn on might be due to a gpio problem.   I had that phenomenon when the battery voltage was not being measured and the under voltage shutdown was triggered.   Check the debug print to see measured battery voltage.

Don’t worry about the warnings… you can add a # include <stdio.h > to config.h if it isn’t there to eliminate the printf warnings.

Chris

http://vamfun.wordpress.com
Robot musing blog
iPhone

s56jsc, 2021-08-11:
OK understand. 
I was thinking to use in speed mode with PID. I did PID in arduino code it work, but not as I wont.
Anyway I'm giving up with 4x hoverboard wheel on mower. It it simple not enough torque for mower.
I will use 24V motor with reductor and PWM. This is what I know well. With this wheel I will build bobby car or similar

BR Simon

closed 2022-01-01 by EFeru

UcanTospa, 2025-01-05:
Hey, I am making a 2 axis robot to shoot at balloons autonoumusly for a competition. I decided to use a hoverboard instead of expensive electronics. I am not really a coder I only know basic stuff, so I did not really understand how to implement uart angle control and didnt understand anything in hoverserial.ino. Is there a detailed example for that? And also, 4 degrees resolution is not really enough for our project so I want to use an external encoder like AS5600. How do I integrate it?