EFeru/hoverboard-firmware-hack-FOC #140: Nunchuk Control

farev, 2021-02-13:
Hello, I had the nunchuk control working properly and then I updated th firmware and it stopped working. I don't believe my nunchuk is the problem because it was able to work before. I also tried using the code I had used before but is still didn't work. What may be causing this problem?

Candas1, 2021-02-14:
Can you explain a bit more what you mean by not working? Do you get beeps, no beeps?
If even the old firmware doesn't work, are you sure the wiring is OK?

In case you did the Auto-calibration by mistake, you can either completely delete the chip in stlink utility, or redo the calibration.

farev, 2021-02-14:
I do get 4 beeps which means a timeout. I have checked the wiring several times but maybe I am connecting something wrong. I am using the right sensor board cables the ground and the two signal wires and I am using the 3.3v pin.

Candas1, 2021-02-14:
Just to be sure, you are using nunchuk variant right? Not another variant with nunchuk as auxiliary input

farev, 2021-02-14:
Yes I am using nunchuck variant. And should there be some current passing through the nunchuck because I measured with a voltmeter one end on the 3.3v pin and the other one in the right sensor board ground and I didn’t get nothing.

Candas1, 2021-02-15:
It could be that your nunchuk is not working anymore ? can you get a spare one ?

farev, 2021-02-19:
I was able to get a spare nunchuk and I tried it but it still didn't work.

farev, 2021-02-20:
I tried the USART variant to control it with an arduino and it didn't work either. I got 3 beeps.

Candas1, 2021-02-20:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC#diagnostics

Please check the wiring.

farev, 2021-02-20:
I did a full chip erase and I flashed it again and now the USART variant works. I tried the nunchuk variant and it still didn't work.

farev, 2021-02-20:
I am modifying the arduino code to be able to use a nunchuk connected to the arduino and using the USART variant. What is the max speed I can go, the test speed is 300 so I don't what is the max value it can go.

Candas1, 2021-03-01:
Hi,

Range for Steer and speed is -1000 to 1000.

farev, 2021-03-09:
I haven't tested it yet but it should work. After I test it if it work I will close the issue.

farev, 2021-04-12:
It is working great!

closed 2021-04-12 by farev