EFeru/hoverboard-firmware-hack-FOC #141: Connecting with the Hoverboard via ST-Link V2 does not work

royrobotiks, 2021-02-15:
First of all, thanks for the great work on the hoverboard firmware! I'd like to flash my hoverboard with your firmware in order to control the motors via UART. 

A first try ended already in a sparkling mainboard with smoke. So I ordered a new mainboard, put it in the hoverboard and the hoverboard works again after the replacement. However, I still want to flash another firmware on it and I'm still stuck at the very beginning. I have no clue what might be the problem. I'm also a complete STM32 n00b and Windows10 user, for that matter. Any advice will be highly appreciated! 

Here's what I've done so far: 

- soldered a red wire to the main board's 3.3V connection
- soldered a green wire to the main board's SWCLK 
- soldered a black wire to the main board's GND
- soldered a blue wire to the main board's SWDIO

Then I connected SWCLK, SWDIO and GND with the ST-LINK V2. 3.3V (red wire) is NOT connected.

After that I followed the instructions on https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/How-to-Unlock-MCU-flash - "How to Unlock MCU Flash". And I tried to connect with the controller while the board was powered on via its own battery.

Unfortunately this did not work. The error message was always "can not connect to target!". It asked me to select 'Normal' or 'HotPlug' mode from Target->Settings menu. So I've tried all options but nothing worked. 

I hope anyone can help me. Thanks a lot in advance!

Here are some photos of my setup:

[image]

[image]

[image]

[image]

In case you need more detailed info, please let me know.

Candas1, 2021-02-15:
Hi, 
Unfortunately this firmware is not compatible with this mainboard variant.
It has a totally different chip and design.

Here is another firmware you can use:
https://github.com/cloidnerux/hoverboard-firmware-hack

royrobotiks, 2021-02-15:
I'd like to control the two motors via UART. Ultimately, I want to build a line following robot with a raspberyPi & camera, which can move people. 

Thanks a lot for the link! I'll give it a try tomorrow.

closed 2021-02-16 by royrobotiks