EFeru/hoverboard-firmware-hack-FOC #142: Question: USART Method

Nejat11m, 2021-02-16:
Hello I am using the USART Method with an arduino nano. My question is what the speed value ist or which unit it has. For example if I write Send(0, 100) then I get a the serial monitor around 32rpm. what is the number 100. What does this number indicate? 
And is there a possibility to send with the Send() function the rpm?

Candas1, 2021-02-17:
Hi, I already told you in the group if you use speed mode then the command you send is in rpm.

Nejat11m, 2021-02-17:
> 
> 
> Hi, I already told you in the group if you use speed mode then the command you send is in rpm.

Hello! I tried to use speed mode but it does not work. I changed in the Firmware in config.h #define CTRL_MOD_REQ    VLT_MODE to #define CTRL_MOD_REQ    SPD_MODE. Is that enough to have the speed mode?

After I changed this and uploaded the firmware I tried the arduino Code but the wheels did not move. I wrote in the arduino code Send(0,50). 

Do you know how I can solve this issue?

Candas1, 2021-02-17:
Just changing from VLT_MODE to SPD_MODE shouldn't be an issue.
You need to check your setup.

EmanuelFeru, 2021-02-17:
How do you send the data from the Arduino? Because it has to have a specific structure. Have look at the Arduino example in the repo.

s56jsc, 2021-02-28:
Hi there
I have more problem with UART mode. My friend prepare me code for STM32 and I successful uploaded it. I attach it to my PC and I got data from controller.
[image]
But when I touch command it hang up. I must remove battery from controller. ?!?!
OK then I upload your arduino program (on my arduino UNO) and connect it on. Now I don't got nothing from controller. I connect it to pin 2 and 3 as de script. Nothing in nothing out.
[image]
What do I do wrong? What command are used for serial?
About my project. I'm doing 4x4 mower. Front and real wheels are not same size so I need arduino board to  calculate this difference. 
Also when I stop mower wheels must be blocked and breake when I go down from hill.
I will be very helpful if someone compile me file for controller

best regards
Simon

Candas1, 2021-03-01:
Hi,

If it's hanging up it means there is a hardfault in the firmware.
What have you modified in config.h ?
Are you using USART Variant ? Why are you use DEBUG output ?

s56jsc, 2021-03-01:
Hi
It hangup when something wrong with UART data. If I don't attach cable this is not happens. Look like bug in FW.
I don't know. My friend did it. I only ask him to make me HEX file for UART.
if I open DEBUG I receive only numbers. So something come throw.

Candas1, 2021-03-01:
OK but I need the details to help.

s56jsc, 2021-03-01:
Thanks for your help! Because I'm stack and I don't know what else to do...
This I got if I open DEBUG on arduino
[image]
About how was compiler STM32 file I really don't know. So I will be very helpful if someone send it to me, so I know that is OK

EmanuelFeru, 2021-03-01:
@s56jsc Simon, the Arduino example needs the UART Variant to generated on the mainboard. I guess/hope your friend built the correct variant for you. The on the Arduino:
- what Arduino type do you have?
- and which pins do you use for the Software Serial? I the example I used: Rx = pin 2, Tx = pin 3
- My guess is that the either the wiring Tx/Rx from mainboard to Arduino is not ok. Or the pins you selected on the Arduino for the Software Serial are in conflict (not all pins on the Arduino can be used for Software Serial)
- can you provide a schematic or a picture with your wiring? Then, we can maybe see the issue.

s56jsc, 2021-03-01:
I have arduiono UNO and I use pin 2 and 3 as descript in program. Program is not modificate.
Also for me is little strange why it not work, but on PC work well. On oscilloscope I can see signal. 
Also wiring is nothing special. GND to GND, RX to TX, TX to RX.
Here is picture of my boards.
[image]

s56jsc, 2021-03-02:
After long, and hard time I got some progress.
I will explain if someone gave same problem, because this was not de-scripted or I didn't note it.
I have PCB with GD32F103. On this MPU **UARTS are not same like STM32F103 !!!!!!**
[image]
[image]
So PB10, PB11 in not UART3 (on right side). UART3 is on position PC10, PC11 on **left side and they are 5V tolerant!**
[image]

Now I'm looking what command are. I know only Send (0, speed_of_motor);
what about left 100, right 200 and disable motor, etc ???

Nejat11m, 2021-03-04:
> 
> 
> How do you send the data from the Arduino? Because it has to have a specific structure. Have look at the Arduino example in the repo.

I used the Arduino example but it did not work in speed mode. I switched back to voltage mode and now it works again. Do you may know how the speed values ​​and the motor output values ​​(rpm) are related in voltage mode?

Nejat11m, 2021-03-06:
> 
> 
> Just changing from VLT_MODE to SPD_MODE shouldn't be an issue.
> You need to check your setup.

I only changed it form VLT_MODE to SPD_MODE and then notihng worked. The wheels did not rotate. After that I changed it back to VLT_MODE and then everything worked again.

Candas1, 2021-03-06:
To be honest it should both work. 
Speed mode should work even with smaller values as it's a rpm command.

What values are you sending?

Nejat11m, 2021-03-06:
> 
> 
> To be honest it should both work.
> Speed mode should work even with smaller values as it's a rpm command.
> 
> What values are you sending?

I tried different values. I tried for example Speed 50 Steer 0, Speed 100 Steer 0 and some other values.

Candas1, 2021-03-06:
Just to clarify, the firmware expects initial values to be in - 50 to 50 range for both steer and speed at startup to enable the motors, then you will hear beeps. This is for safety.

EmanuelFeru, 2021-03-09:
@Nejat11m is it working? Then please close the issue. Thank you!

closed 2021-05-12 by Candas1