EFeru/hoverboard-firmware-hack-FOC #143: VARIANT_USART commands

diogoc, 2021-02-18:
Hello. It would be nice if it were possible to send other commands in variant_usart, such as the mode and type of motor control.

Candas1, 2021-02-19:
It's actually possible but it's not documented.
If you open this [web tool](https://candas1.github.io/Hoverboard-Web-Serial-Control/), you can see I am using it already in Ctrl view with protocol hovercar.
[image]

If you use sideboard_usart instead of control_usart, you can send following command:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/36953c6fe9ba90593d21150113459b0a4bb4e312/Inc/util.h#L47

- start : 0xABCD
- pitch (not relevant here, you can sent 0)
- dpitch (not relevant here, you can sent 0)
- steer : -1000 to 1000
- speed : -1000 to 1000
- switches
  - 0:OFF or 1:ON (for switching input when dual input is used)
  - Control type 00:FOC 10:Sinusoidal 01:Commutation
  - Control mode 00:Voltage 10:Speed 01:Torque
  - Field Weakening 0:off 1:on
  - 0000000000
- checksum

This you can adapt in the arduino code

diogoc, 2021-02-19:
Thank you. I will try.

closed 2021-07-04 by EmanuelFeru