EFeru/hoverboard-firmware-hack-FOC #145: Having problem with breaks hovercar

mochiianune, 2021-02-24:
Hi im using the variant hovercar. Got problem with the breaks everytime that i used the break insted of making the wheels to stop or to slow it down it go realy fast. Im making the test run with all the wheels is on the ground and im riding the hovercar. And I did activate the reverse wheel turn. Thanks

Candas1, 2021-02-24:
Hi. 

How is your setup looking like?
Do you use pedals?
Do you have a video of how it behaves?

mochiianune, 2021-02-24:
Yes I put to pedals on it. I dont have a video but everytime that I step on the pedal break instead of going slow or stop the wheels turn fast

Candas1, 2021-02-24:
But throttle works well?

mochiianune, 2021-02-24:
Yes throttle works really well.

mochiianune, 2021-02-24:
And reverse go really well also when I step on the break twice

Candas1, 2021-02-24:
Are you sure your wiring is ok ?

EmanuelFeru, 2021-02-24:
You can check the Hovercar schematic here: https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/Variant HOVERCAR

From the symthoms can it be that the Brake potentiometer is damaged?

mochiianune, 2021-02-24:
It might be the pot I was suspecting thesame thing coz my wiring was all correct. Im going to switch the pot beetween the break pot and the throtle pot later on

EmanuelFeru, 2021-03-09:
Fixed?

mochiianune, 2021-03-20:
Sorry for late reply. I did found the problem it was my motor hall sensor inside the motor. When i change my motor hall sensor everything is running good

mochiianune, 2021-03-20:
[image]

EmanuelFeru, 2021-03-20:
Good that you found it. Then I will choose this topic.

closed 2021-03-20 by EmanuelFeru