EFeru/hoverboard-firmware-hack-FOC #147: UART command (arduino)

s56jsc, 2021-03-02:
How to use arduino command?
Command is Send(steer,speed)
Speed understand. But what is steer? It make no different if I put 0, 1, -1 or 10.
How to drive different left and right wheel?

Candas1, 2021-03-02:
Hi,

Both steer and speed are values between -1000 to 1000

EmanuelFeru, 2021-03-09:
@s56jsc all working, then you can close the issue.

s56jsc, 2021-03-09:
Yes it work. I only missing command for enable/disable driver. But no problem work also like this!

s56jsc, 2021-03-21:
Hi guys
My boards now working well https://youtu.be/uJ8dwduxY_M
But I found new problem. I cannot read data from controller.
If I use only one virtual port then it work (with some error, but work), but I use 2. virtual serial no go!. I found also limit from arduino site
[image]
How can reduce serial speed on hoverboard driver? I try this, but not work well
[image]

EmanuelFeru, 2021-03-21:
Nice robot! Just wondering there is no slip because you use bigger wheels in the back?

I think the uart limitation is from Arduino side. The location of the baud rate change is ok. If it is the same on the Arduino side, it should work.

s56jsc, 2021-03-22:
No, I have calculation and wheels in back are rotating slowly then at front.
And yes, it is limit from arduino site. This is reason, that I wont to decrist bourdate. If enter 9600 here something strange happens with controller. It not work smoothly. I will do some extra tests.

EmanuelFeru, 2021-03-22:
What is the behavior you have? Does it beep now and then? Maybe you need to increase the timeout. It could be that it enters into timeout because the flow of UART data is less, so it is more prone to timeout.
You can increase it here:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/36953c6fe9ba90593d21150113459b0a4bb4e312/Inc/config.h#L605

s56jsc, 2021-03-22:
I found problem. It was in my arduino code.
But for reciving code from controller... this is other story. With one ony virtual port I receive few packet the I receive some invalid packet then nothing. Also with osciloscope I note, that nothing receive from controller. I will keep working on this

closed 2021-05-12 by Candas1

abdelrahman1234567, 2022-01-20:
Hi,
I'm having the same problem now because I don't know how to control each motor independently. For example, I have two motors the first one I want it to move with speed -100 and the second one with speed 100 (move right or left) how can I do such a thing in the Arduino code?

Candas1, 2022-01-20:
Hi,

It's not the same problem, if you have a different question you should open a new question.
For tank/squid steering, you have to change the firmware itself, check what this guy did in his [code](6132b2bc4256fb91e286567267c60e01209d9677) for example