EFeru/hoverboard-firmware-hack-FOC #148: Configuration code for 24V

johnnybreizh, 2021-03-05:
Hi team.
Great job for all ! Thanks a lot !
I try to change some values in configuration code for use 24V battery.
But it still cut power in safety mode i guess.
So can you indicate me how i have to process to have a 24V compatible code ?
Is it possible ?
Thank you for your precious help.

johnnybreizh, 2021-03-05:
Because i measured the motherboard stay 8sec on then power off by itself.
maybe be its not due to my 24V power, maybe its other reason ?

johnnybreizh, 2021-03-05:
Ok i search here and i do that : 
"To disable the shut down due to battery, search in main.c for poweroff() and remove the instance where the check for low battery voltage is done."
So its okay now.
Have great day !

EmanuelFeru, 2021-03-06:
That is one way to prevent it but then you will not be protected for low battery. Have a look in config.h and update the battery parameters to your battery, mainly number of cells and voltage levels. Also, don't forget to do the battery calibration.

Candas1, 2021-03-06:
It's also documented in the

closed 2021-03-09 by EmanuelFeru