EFeru/hoverboard-firmware-hack-FOC #151: hovercar input not working

hunty1987, 2021-03-22:
im struggling to get this firmware working. i have been using the bbcar firmware and that installs and works great just dont like the controls. so i decided to try this one. but i cant get the pedals working. i have tried calculating the adc values and that didnt work. i have tried auto calibration but there is no 2nd beep when holding down the power button. to i need to uncomment something for the auto calibration to work or is it supposed to work straight away.
all i have done to the firmware is choose my variant and set adc values. am i missing something?

EmanuelFeru, 2021-03-22:
Auto-calibration is on by default. You need to release the power button after power on and then press and hold to enter autocalibration.
Did you also check that schematic is ok as here:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/Variant-HOVERCAR

hunty1987, 2021-03-22:
thats where im going wrong then thought it was hold it as you turn it on. works like a charm now cheers for your help

EmanuelFeru, 2021-03-22:
Good to hear. Enjoy! ;-) I will close the issue then.

closed 2021-03-22 by EmanuelFeru