EFeru/hoverboard-firmware-hack-FOC #152: 4WD question

interlark, 2021-03-22:
Hi! Very interesting project.
Is it possible to control 2 hoverboards with one arduino uno? And if it is so, what's the solution to synchronize all 4 wheels, any suggestions?

EmanuelFeru, 2021-03-22:
Depends how are you controlling it and what is your input. The idea is to have both mainboards receive the same input. Just some examples (didn't try myself):
- if you used ADC as input, then you have to make the 2 mainboards GND common, and connect the potentiometer(s) output to both mainboards ADC input. Take 3.3V from one board only.
- if you use UART, then send the same data to both boards, again common GND.

interlark, 2021-03-22:
Thanks for reply. Okay, so it doesn't matter whether I place wheels like (HWWx = x-th hoverboard's wheel):
```
[HWW1 LEFT] - [HWW1 RIGHT]
[HWW2 LEFT] - [HWW2 RIGHT] 
```
or
```
[HWW1 LEFT] - [HWW2 RIGHT]
[HWW1 LEFT] - [HWW2 RIGHT] 
```
?

EmanuelFeru, 2021-03-23:
It does matter. You can use the arrangement you prefer, just make sure to comment/un-comment in `config.h` the following defines to match your setup:
```
 #define INVERT_R_DIRECTION                // Invert rotation of right motor
 #define INVERT_L_DIRECTION                // Invert rotation of left motor
```

interlark, 2021-03-23:
@EmanuelFeru Thank you, mate!

closed 2021-03-23 by interlark

SchO2n, 2023-03-28:
> Depends how are you controlling it and what is your input. The idea is to have both mainboards receive the same input. Just some examples (didn't try myself):
>
> * if you used ADC as input, then you [...]

Hello, can we link the two GND of the motherboards if they are powered by two different batteries? Tks