EFeru/hoverboard-firmware-hack-FOC #153: remove pwm_margin to get 100% PWM

Rudi84, 2021-03-24:
Hi Emanuel,

is it possible to adapt the current measurement to allow 100% PWM. If I am correct currently we are clamped at 89%. With this also the sin3 or sin shape is destroyed at higher input values. In practice this does not matter that much, because inductance of motor seems big enough to smooth this. Getting 100% would have following advantages:

- get real harmonic shape also for high input
- gain 11% of speed
- for FOC also starpoint shift (like sin3 in Sin-mode) can be implemented, which gain theoretically additional 14% more speed and with some clever starpoint shift strategy power losses in FETs can be reduced

Unfortunately I do not the meas equipment by my self and I know this is not a quick fix. But maybe if you have some time and interest, can you try to modify to get 100% PWM running?

Thanks and best regards
Rudi

EmanuelFeru, 2021-03-24:
Hi Rudi,

Thank you for the feedback. You are right regarding the shape beeing clipped above 89%. For Commutation and Sinusoidal I see no issue to go to full 100% and I will consider it in the next commit. However, for FOC i tried many things to improve current measurements, but they are very unreliable at high PWM values. So i am afraid we cannot increase the max PWM for FOC. Maybe implementing current observers, e.g., Kalman filter can improve the current measurements. But unfortunately at the moment I cannot spend more effort on it.

Rudi84, 2021-03-30:
Hi Emanuel, 

thanks for already updating this. "Unfortunately" I am using you FOC-Torque Mode because, you can perfect control the throttle and limit current to not over stress the battery. But I am also using the modified delta motor, so I am also not sure if e.g. a tuned Kalman filter for your motor work on mine. So can be closed for now

Rudi84, 2021-04-19:
Hi Emanuel,

I tried to understand the current measurement problem. From Hardware point of view current is measured over LS Fet, so problem appears if LS Fet is off or on to short, so at high PWM. But it should be no problem at low PWM? So in theory, low margin can also be removed? This would already make half the way.
Best regards

Candas1, 2022-07-04:
I think this issue can be closed, I am investigating other ways of increase PWM with FOC but it will take a lot of time.

closed 2022-07-04 by Candas1

Rudi84, 2024-05-13:
Hi,

I finally found some time to try this out. Means shifting midpoint to bottom (bottom clamp) and removing the lower margin. The measured result is more or less like expected:

+ less thermal losses in LS FET --> only 80% thermal power compared to mid clamp
+ ~2% more no load speed (should be in theory 5%)

- motors seems to run bit more noisy, not a lot, but recognizable a t low speed

Conclusion: If you do not want to squeeze out the last % of performance and do not have a problem with cooling your electronic, keep the implemented mid clamp version

Candas1, 2024-05-13:
Hi,

That's also called null_v0, it's being discussed [here](https://github.com/vedderb/bldc/pull/590) for VESC.
The reason for the noise could be the deadtime, for a small pwm duty cycle the mosfet might not even have time to open.