EFeru/hoverboard-firmware-hack-FOC #155: PWM version - decrease power

lukis01, 2021-03-29:
Dear Emanuel thank you very much for your work and contribution you've done so far. As being total noob in this field of experience, I would like to ask you for some advice.
After couple hours of learning about Git, procedures, some videos I finally managed to flash mainboard via PlatformioIDE and got working RC connection.
The only thing I would like to solve is decreasing of  power - for my project I'd like to get much less speed (lets say about 7 km/h for full throttle) than available now.
Do you think you could point me to solution please?
Bests
Lukas

Candas1, 2021-03-30:
Hi,

You can reduce max speed in config.h (N_MOT_MAX), and reduce SPEED_COEFFICIENT not to hit max speed too quickly and huge full throttle range.

lukis01, 2021-03-30:
great...I guess I should also manipulate I_DC_MAX value to get I_DC_MAX = I_MOT_MAX + 2A. Am I correct? Thanks a lot

lukis01, 2021-03-30:
perfect.thx a lot

closed 2021-07-24 by EmanuelFeru