EFeru/hoverboard-firmware-hack-FOC #160: default config.h will ERROR: DEBUG_I2C_LCD and SERIAL_USART3 not allowed.

RoboDurden, 2021-04-04:
I think should not be active by default. Will cause 
`#error DEBUG_I2C_LCD and SERIAL_USART3 not allowed. It is on the same cable.`

```
// ############################### DEBUG LCD ###############################
#define DEBUG_I2C_LCD                // standard 16x2 or larger text-lcd via i2c-converter on right sensor board cable

```

Candas1, 2021-04-04:
Hi,

DEBUG_I2C_LCD is not active by default. 
Is it what you mean?

RoboDurden, 2021-04-04:
Sorry yes indeed, it is not active: 

https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/0709e4cb62ef1d0bfedfc6bc3187614d5862acd6/Inc/config.h#L259

For some strange reason, my onlinie compiler fetched a version with DEBUG_I2C_LCD active.
Sorry for the confusion.

closed 2021-04-23 by Candas1