EFeru/hoverboard-firmware-hack-FOC #162: VARIANT_USART command dropouts

vamfun, 2021-04-16:
VARIANT_USART command dropouts 
Hi.. I am trying to control a Hoverboard H1 with an ESP32 .   I need the left ADC inputs for encoders so I’m trying to make a I2c or usart serial connection.   I was unable to drive the STM32 as a slave with the ESP32 master...see issue #158.    
So next I decided to try the VARIANT_USART with usart3 as input.   
I was able to get the Esp32 to run the Arduino test code with slight modifications to drive my FOC PID servo variant.    I basically commanded  a full rotation in 200 steps and repeated.  The feedback was ok but I randomly dropped a few command steps every rotation.    
I thought it might be checksum errors or a circular buffer code error but analysis showed that there were no checksum errors only partial data transfers.
I instrumented the uart3_check_rx () to print out old and present positions in the buffer.   Data dropouts occurred when there were less than 8 bytes differential between old (o) and new positions (pos) as shown in the figure.

￼[image]

I tried slower and faster BAUD rates  and driving with Arduino UNO using the repository Arduino code  but problem persists.

I next decided to try usart2 but got nothing but serial time outs because the usart2 rx was not reading the Esp32 commands.

So I decided to write STM32 uart - dma test code from scratch using STM32CUBEMX .   It basically reads keyboard input  data from a terminal and echos the inputs back through the USART tx .  The code can test usart3 and usart2.    The usart3 works fine as I suspected but the usart2 rx doesn’t function.   This is strange since the ADC inputs work ok.

Q1:  has anyone achieved stable uart3 data transfers?   If so are there any hardware connection tricks to lessen  the noise other that keeping the wires short.  My connection wire length is basically the length of the hoverboard right input cable.

Q2:  has anyone successfully used ESP32 dev board with the VARIANT_USART?  If so can someone try my FOC servo code (see issue #138)  with an Esp32 with their board.   I suspect I may have a damaged board.   I can post my ESP32 code and also my STM32CUBEMX uart test code if required.

Q3: any ideas on what I can try next?
Thanks chris

Candas1, 2021-04-16:
Hi, sorry I didn't reply about I2C because I never worked with it.

With general usage of this firmware dropping some usart frames is not really a problem as we get the right information in the next frame.

Because of that there is no acknowledge mechanism. 
Wrong frames are just rejected if:
- command length is not correct
- start frame is not correct
- checksum is not correct

What can happen also is if you have a delay somewhere when sending the frame.
Idle line detection interrupt is triggered and the frame could be read as 2 frames.

vamfun, 2021-05-01:
Thanks @Candas1
I did a lot more looking into why these data drop outs were occurring.  I'm sure it's coming from the idle interrupt.  Why I don't know yet...but I believe it's on the STM side rather than the Esp32.  
I am sending commands at 5 ms now and modified the usart3_rx_check(void) function to allow two frames to receive 8 bytes before rejecting data.    So now I get fairly smooth control and no lost frames.

The spurious double frames occur every 1.239 seconds on average   7 bytes are xfered in first frame and remaining 1 byte 1 ms later.  

I noticed that there is a slight change in buzzer frequency when I run at 5 ms command rate.  Possible interrupt contention.

I'll post picture and video.

Also tried using dmarxcpltcallback to trigger the processing rather than idle interrupt but just got usart time outs.  

Still big mystery why this is happening.[image]

EmanuelFeru, 2021-05-01:
@vamfun how did you configure 2 frames for the idle line interrupt? Because i actually i prefer 2 frames instead of 1 for robustness and to be sure the line is for sure idling meaning the data transfer is finished.

vamfun, 2021-05-02:
here is the code .......i commented out your code and basically added a counter of  frames.   If 8 bytes were received within two frames then I call the decoding, otherwise the frames were rejected.    I print out anytime it takes two frames to get 8 bytes and measure the time between prints.

```
`
void usart3_rx_check(void)
{
  #if defined(DEBUG_SERIAL_USART3) || defined(CONTROL_SERIAL_USART3) || defined(SIDEBOARD_SERIAL_USART3)
  static uint32_t old_pos;
  uint32_t pos;  
  pos = rx_buffer_R_len - __HAL_DMA_GET_COUNTER(huart3.hdmarx);         // Calculate current position in buffer
  #endif

  #if defined(DEBUG_SERIAL_USART3)
  if (pos != old_pos) {                                                 // Check change in received data
    if (pos > old_pos) {                                                // "Linear" buffer mode: check if current position is over previous one
      usart_process_debug(&rx_buffer_R[old_pos], pos - old_pos);        // Process data
    } else {                                                            // "Overflow" buffer mode
      usart_process_debug(&rx_buffer_R[old_pos], rx_buffer_R_len - old_pos); // First Process data from the end of buffer
      if (pos > 0) {                                                    // Check and continue with beginning of buffer
        usart_process_debug(&rx_buffer_R[0], pos);                      // Process remaining data
      }
    }
  }
  #endif // DEBUG_SERIAL_USART3

  #ifdef CONTROL_SERIAL_USART3
  uint8_t *ptr;	int cnt = 0; int sum = 0;
	
  if (pos != old_pos) {                                                 // Check change in received data		
		ptr = (uint8_t *)&commandR_raw;                                     // Initialize the pointer with command_raw address
//    if (pos > old_pos && (pos - old_pos) == commandR_len) {             // "Linear" buffer mode: check if current position is over previous one AND data length equals expected length
//      memcpy(ptr, &rx_buffer_R[old_pos], commandR_len);                 // Copy data. This is possible only if command_raw is contiguous! (meaning all the structure members have the same size)
//      usart_process_command(&commandR_raw, &commandR, 3);               // Process data
//     // printf("call1 \r\n");
//    } else if ((rx_buffer_R_len - old_pos + pos) == commandR_len) {     // "Overflow" buffer mode: check if data length equals expected length
//      memcpy(ptr, &rx_buffer_R[old_pos], rx_buffer_R_len - old_pos);    // First copy data from the end of buffer      
//			if (pos > 0) {                                                    // Check and continue with beginning of buffer
//        ptr += rx_buffer_R_len - old_pos;                               // Move to correct position in command_raw
//        memcpy(ptr, &rx_buffer_R[0], pos);                              // Copy remaining data
//     
//       // printf("buffer overflow pos>0 \r\n");  
//			}
//      usart_process_command(&commandR_raw, &commandR, 3);               // Process data
    // **********vamfun rewrite that allows two reads to occur before ignoring data**************
	 if (pos > old_pos ) {             // "Linear" buffer mode: check if current position is over previous one AND data length equals expected length
      memcpy(ptr, &rx_buffer_R[old_pos], pos-old_pos);                 // Copy data. This is possible only if command_raw is contiguous! (meaning all the structure members have the same size)
      //usart_process_command(&commandR_raw, &commandR, 3);               // Process data
      sum += pos-old_pos; 
    } else                                                             // "Overflow" buffer mode: check if data length equals expected length
      {
				memcpy(ptr, &rx_buffer_R[old_pos], rx_buffer_R_len - old_pos);    // First copy data from the end of buffer      
			if (pos > 0) {                                                    // Check and continue with beginning of buffer
        ptr += rx_buffer_R_len - old_pos;                               // Move to correct position in command_raw
        memcpy(ptr, &rx_buffer_R[0], pos);                              // Copy remaining data
       
        
			}
			sum += rx_buffer_R_len-old_pos + pos   ;
				
 		}
			
		  if(sum ==8 && cnt<2 )
			{usart_process_command(&commandR_raw, &commandR, 3);
			if (cnt == 1) {printf("%d ",commandR.speed);printf("p-o %d \r\n", pos-old_pos);}
			cnt = 0;}
      else 
			{
      if(cnt==0) printf("cnt 0 %d %d ms %d\r\n", pos-old_pos, commandR.speed, buzzerTimer/16-ms_last);
			if (cnt>0 )printf("err %d \r\n", pos-old_pos);
			cnt++; ms_last = buzzerTimer/16;
		}
  }
  #endif // CONTROL_SERIAL_USART3
```

vamfun, 2021-05-02:
Accidentally closed this issue...sorry..   i think i reopened it.

goosst, 2021-05-10:
I just started using the firmware today (so far so good, flashing went fine) and wanted to use an ESP32 or ESP8266 to send and read serial commands. (Sounded more reliable than using an ADC or PWM.)
Do I understand correctly there are still some issues deeply embedded in the STM firmware? or would there already be a branch somewhere with some experimental working version?

EmanuelFeru, 2021-05-10:
The usart variant works, you can use it. Vamfun implemented additional position control and there he experienced some drop outs.

closed 2022-03-06 by EFeru