EFeru/hoverboard-firmware-hack-FOC #164: VARIANT_HOVERCAR_Field Weakening

Jenya-m, 2021-04-22:
I am using a field weakening of 7-9 amps. The maximum speed has become noticeably higher, but there is a nuisance in controlling the hovercar. After accelerating to the maximum speed (30 km / h), if you slightly lower the gas or completely, there is a sharp braking and a drop of speed to 20 km / h (maximum speed without weakening the field) and then coasting as usual.

###### MOTOR CONTROL #####
#define CTRL_TYP_SEL FOC_CTRL
#define CTRL_MOD_REQ TRQ_MODE
#define I_MOT_MAX 23
#define I_DC_MAX 25
#define FIELD_WEAK_ENA 1
#define FIELD_WEAK_MAX 9
#define PHASE_ADV_MAX 25
#define FIELD_WEAK_HI 1500
#define FIELD_WEAK_LO 800
####### VARIANT_HOVERCAR SETTINGS ####
#define CTRL_MOD_REQ TRQ_MODE

And one more thing: with an attenuation of 8-9 amperes. If the brake is held to a full stop, then at the end of braking it begins to throw off the pilot with sharp jerks back and forth, as long as he holds the brake.

Without weakening this glitch, there is not even 25 Amperes!

the2jakes, 2021-04-23:
Hi Jenya-m, if it can be of any help finding a solution, I have the same issue you describe. I use the Skateboard variant, and when Field Weakening was enabled I had this sudden drop in speed when releasing my remote control trigger from max to middle (0) position. Even a small trigger release of a few mm can generate this behavior. I noticed also that it affected only the front right wheel of my mountainboard, (I fell of my board, and broke the front truck).

Here are the settings I had for field weakening:
#define FIELD_WEAK_MAX  5
#define PHASE_ADV_MAX   25
#define FIELD_WEAK_HI   1000
#define FIELD_WEAK_LO   750

motor settings:
N_MOT_MAX       1200
I_MOT_MAX       25
I_DC_MAX        27

and I use FOC and torque mode.

Since this issue happened, I disabled field weakening. I’m interested in finding a solution, because the extra speed I got was appreciated 😊.

Just to illustrate the mountainboard I’m talking about:
 [HummerBoard video](https://youtu.be/-geRIjU13-0)

EmanuelFeru, 2021-04-24:
I identified why is this happening. The behavior comes from the id_Tgt current which is the one responsible for Field weakening. The id_Tgt is a function of the Throttle pedal and as the throttle is released the id_Tgt becomes 0. However, if the actual motor speed is high, the iq controller cannot realize 0 current (freewheeling) with id_Tgt also 0. It needs some higher id_Tgt to realize 0 current through the motor.

I don't have yet a solution available. As a mitigation, I recommend not to release the Throttle immediately when operating in the higher rpm where Field Weakening. Wait for the speed to reduce and then release the pedal.

the2jakes, 2021-04-24:
Hi Emanuel, good to know you found the cause 👍, thanx for sharing 😊. Hope the solution will not be too complicated to code 😉.

matan144, 2021-08-07:
Is there any solution yet? I also making my own skateboard and I found this problem with field weakening...

Jenya-m, 2021-08-11:
Я нашёл только один выход - не использовать ослабление поля. 
Я поднял напряжение питания (54 вольт) и в результате получил такую же скорость и большую мощность только без глюка.
--
Отправлено из Mail.ru для Android суббота,

Kowitzki, 2021-12-28:
I have got a fix for that.
Problem:
The problem is that field weakening depends on the throttle signal. If you drive very fast and let go of the throttle, the field weakening goes to 0, but the motors still turn above their max speed. Thus the motor voltage is higher than the battery voltage and a very high current flows back through the body diodes of the mosfets back into the battery.

Solution:
- Not tested with different mode and configurations! - Be careful at trying this! -

If you just use torque mode you can exchange a variable in the file "BLDC_controller.c".
In line 2168 "DataTypeConversion2" can be replaced with "rtP->r_fieldWeakHi".
By that the dependency on the throttle input is overwritten.

the2jakes, 2021-12-29:
@Kowitzki thanx ! 👍. I’ll give this a try (scary try though…).

So all I need to do is replace the variable « DataTypeConversion2 » by « rtP->r_fieldWeakHi » in line 2168 ?
(and of course activate properly field weakening option in config.h).

the2jakes, 2021-12-29:
@EFeru , hi, what do you think about Kowitzki’s code modification for this field weakening deceleration issue ?

EFeru, 2021-12-30:
I didn't check yet his fix. If you try it is at your own risk... Maybe you can just run the motor in the air first.
I will have to check the fix myself but right now i have other higher priorities in the pipeline unfortunately.

NikasAl, 2022-01-06:
> If you just use torque mode you can exchange a variable in the file "BLDC_controller.c". In line 2168 "DataTypeConversion2" can be replaced with "rtP->r_fieldWeakHi". By that the dependency on the throttle input is overwritten.

I tried this fix today. Spun the wheels in the air and the braking effect when the gas was released was gone. Thanks! I will test it on the track when the opportunity arises.

EFeru, 2022-01-07:
@NikasAI thanks for trying it. It is a very good input for us. I hope i have some time soon to take a closer look and if possibly fix it and generate new code for it.

EFeru, 2022-03-06:
I implemented a fix for Field weakening. @NikasAl @Kowitzki @the2jakes if you guys can give it a try, it would be great. Thanks!

Jenya-m, 2022-03-15:
Hello. Today I tested a new source, everything went well.

#define MOTOR_LEFT_ENA                  
#define MOTOR_RIGHT_ENA              
#define CTRL_TYP_SEL    FOC_CTRL     
#define CTRL_MOD_REQ    TRQ_MODE        
#define I_MOT_MAX       25           
#define I_DC_MAX        27       
#define N_MOT_MAX       1700           
#define FIELD_WEAK_ENA  1               
#define FIELD_WEAK_MAX  7         
#define PHASE_ADV_MAX   25             
#define FIELD_WEAK_HI   1000            
#define FIELD_WEAK_LO   750

closed 2022-03-18 by Jenya-m

Kowitzki, 2022-05-04:
I tried the new fix yesterdy. Sometimes the board turns of, if you brake. A friend of mine has encountered the same problem.
With my dirty fix here (https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/164#issuecomment-1001983980), that problem never occured to me.

Jenya-m, 2022-05-04:
I heard about this, but the board and firmware were useless. There was a battery protection.
due to recovery. If the battery is 100% charged, pressing the brake will recharge the battery and shut it off.

Я слышал про такое, но плата и прошивка дыли непричём.  Там  срабатывала защита у батареи.
Из-за рекуперации. Если батарея заряжена 100%, при нажатии на тормоз происходил перезаряд батареи и она отключалась.

the2jakes, 2022-09-16:
Hi @EFeru , thanx a lot for this bug fix, I finally took the time to upgrade my HummerBoard (4x4 hoverboard powered eMountainboard), with the latest FOC firmware version you did: Field Weakening works great ! Vmax was 35km/h before, and now 50km/h.

Thanx again ! 😉👍

[image]

larsmm, 2023-04-01:
The fix broke field weakening for VLT_MODE. 

I created an issue: https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/392