EFeru/hoverboard-firmware-hack-FOC #166: Variant USARt et ESPHOMe

BSD31, 2021-04-24:
Bonjour et félicitation pour cet excellent travail.
J'ai utilisé la variante ADC pour piloter la carte de mon hoverboard et ça fonctionne parfaitement.
J'ai maintenant un autre projet avec un ESP32 et ESPHOME.
ESPHOME permet de programmer facilement des ESPxx et intègre automatiquement les données dans la solution domotique open source Home Assistant.
Auriez-vous déjà travaillé avec ESPHOME et auriez vous un exemple de code pour faire fonctionner l'hoverboard directement depuis Home assistant ?
Merci e votre aide.

Candas1, 2021-04-24:
Hi, 

I haven't heard about anyone using this firmware with ESPHOME.
You can control the firmware through an esp32 with usart variant, there are details in the wiki.

BSD31, 2021-04-25:
Thank you for that answer.
I'm looking on my own but if anyone has already implemented this firmware with ESPHOME, that would be great.

Candas1, 2021-04-25:
If you join the telegram group linked in the main page you can ask to the community.

BSD31, 2021-04-25:
Excellent idea. 
I just joined the group. 
I will put my question to it immediately.
Thank you

closed 2021-05-12 by Candas1