EFeru/hoverboard-firmware-hack-FOC #167: Non linear throttle in Torque mode

the2jakes, 2021-04-26:
Hi Emanuel, thanx for your great work ! 😉👍

I have an issue, and other people have also noticed the exact same symptoms.

I use the skateboard variant, in torque mode, with a RC remote control and PWM receiver.

I’ve calibrated the throttle range by pushing the motherboard power button for 5sec (several times, and it didn’t solve the issue).

Issue: the throttle isn’t linear at all. At 20% of the remote throttle range I get 100% power at the wheels. Then from 20% to 100% nothing changes (same max speed/torque).

I don’t use Field Weakening anymore since I crashed when releasing the trigger from max to 70% and it suddenly decreased in speed (there is another issue on this, for which you identified what the issue is). But when I used Field weakening, I noticed it triggered itself at arround 90% throttle, which is normal.

It’s difficult to have a smooth ride with only the first 20% of throttle useful, it accelerates to fast with just a few mm finger movement. And strange to have this « useless » 20% to 90%  range on the remote.

I have this exact same behavior with another RC remote and receiver. (Which work smoothly with my RC cars).

Moreover, I need to trim down my throttle range on my remote, otherwise when reaching 95% throttle I get erratic motor rotation (it stops, restarts, bip, as if I was sending a PWM signal that is over the acceptable range for the motherboard.

Same behavior with wheels in the air, and on the ground with load. (I know that with wheels in the air we sometimes get weird behavior with torque mode, so I tested everything by riding my skateboard.

To illustrate what I’m talking about, here’s a video:
(Conditions:
Torque mode
motor settings:
N_MOT_MAX 1200
I_MOT_MAX 25
I_DC_MAX 27

Field weakening disabled.
Remote freshly calibrated in motherboard, WITHOUT reducing the Trim, so you will also see the eratic behavior near 100% throttle).

[Issue video](https://youtu.be/d1k7EzwXB6U) 

I hope you’ll be able to solve this 😉, I’d love to get a smooth ride on my Hummerboard (which allready is a beast).

[HummerBoard Offroad ride](https://youtu.be/-geRIjU13-0) 
 
Regards,
The2Jakes

Candas1, 2021-04-26:
2 ideas to investigate the issue:
- if you disable both motors in config.h and try again, you can check if the beeps are due to your input or to back emf
- if you try with SPEED MODE instead of TORQUE MODE, we can understand if the non linearity is due to your input or to motor control in TORQUE MODE

EmanuelFeru, 2021-04-26:
@the2jakes let's try 2 things:
- can you give a try with default settings for current?
I_MOT_MAX 15
I_DC_MAX 17
I just want to eliminate a possible issue on current calculation.
- and second as Candas mentioned try a different mode Voltage or Speed to see the behavior.

I remember to have 4 wheel drive. Is the behavior of the other set of wheels the same?

The eratic behavior at 100% behaves as interference on reading the pwm from the receiver. This can be checked by debugging the input values that the mainboard is reading.

zulkep, 2021-04-27:
I have the same problem, FOC TORQUE, how do I make the throttle dial smoother? when the current increases, the start and driving become very sharp, about 40% of the throttle, this is a full set of speed, then 40% there is no speed change, and 20% of the scale at the end turns on the field weakening ... Is it possible to turn on the coast in FOC SPEED mode? if it's important, two wheels 350 watts, tried changing I_MOT_MAX 10
I_DC_MAX 13, the start is smoother, but there is also a dip in the middle, I will try different modes and write

zulkep, 2021-04-27:
I just tried to increase the weight of the scooter (I got up on it with the child), the load increased, the adjustment became smoother by about 60-70%, the failure was only at the end of the scale
I_MOT_MAX 13
I_DC_MAX 15

the2jakes, 2021-04-30:
Hi Emmanuel, I've done some tests. 
I think you nailed it when you talked about current calculation. 
I was running with:
I_MOT_MAX 30
I_DC_MAX 35
because with default setting ( 15 / 17A) I was hitting max current limit very time I climbed a small hill, or did a rapid acceleraiton from static position.

Today I flashed the board with the same program but with:
I_MOT_MAX 15
I_DC_MAX 17
When using the Skateboard on the ground, I get a smooth range of acceleration with the remote control (the all remote range I believe). When wheels are in the air it's visualy the same thing as before (wheels spind at max speed since I hit the 20% gas, which is "normal" I think because it's a Torque mode.

BUT: 
- The hummerboard is much less powerfull with these max current settings, (I run a 45V 20Ah 25C battery, so plenty of power here). I hit very easely the Max current limit (I hear current chopping I believe: like a "whining sound") as soon as I do a rapid acceleration.
- The Brake is much less powerfull too (I set it to 450, and it's very powerfull when I set I_MOT_MAX 30 / I_DC_MAX 35).

I did another test, if it's of any help:
I only activated the front wheels of the skateboard. With I_MOT_MAX 30 / I_DC_MAX 35, I noticed that I could use near 40% of the remote control trigger to control my speed. These matches with Zylkep test (increased the load on his scooter). 

This control range has for sure something to do with Current... ;). 
Question: What is the max current I should set on I_MOT_MAX & I_DC_MAX to use the power I have/need, and not kill my motherboards ? (I've done a few very demanding runs at 30 / 35A setting, I didn't have any issue, but maybe it's a bit too high ?

Just to answer your other questions:
- Yes this behaviour is on the 4 wheels (2 seperate motherboards).
- I don't think the eratic behavior over the 100% gaz position is due to interference, since I now shielded every cable+receiver (which solved my interference issue I had at high speed), and I still get this beshaviour if I push the Trim up on the remote.
 When calibrating the remote range, I have no issue at 100% gaz. It's just if I increase the trim on the remote that I get it (so not a real issue for me). I bought a usb debug reader, I'll give it a try.

EmanuelFeru, 2021-05-01:
When you hear the winning noise is indeed the current chopping kicking in. It is nothing to worry about but i can imagine it's not a nice feeling. You can set the I_DC_MAX higher apart from I_MOT_MAX to try to avoid the current chopping.

I will check in the model what max current is allowed from firmware point of view, I remember is 30A, but I will check.

For the board or motor, i heard people running it at 30A, but i am not sure what is the duration. Because some the mosfets are even rated to 75A, 100A or 110A, but is all about the duration. That's what kills them, because the heat sink cannot cool that fast.

mrlunk, 2021-05-02:
To make use of full throttle range on RC transmitter try changing the range in these lines from -1000 and 1000 to -4000 and 4000
Like this: 

` #define PRI_INPUT1            3, -4000, 0, 4000, 100  // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
    #define PRI_INPUT2            3, -4000, 0, 4000, 100  // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section`
    
    This gave me near ful range trhottle on my Transmitter.
    Full range should be -4096 t0 +4096.
    
    Greetz,
    Peter Lunk

the2jakes, 2021-05-05:
Thanx mrlunk, I’ll give this a try.

the2jakes, 2021-05-07:
Guys, thanx Mrlunk, your tip worked ! 👍

I’ve put this:
#define PRI_INPUT2 2, -800, 0, 3000, 100 // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section

First test I had put -4000, 0, 4000 and brake was much too weak, I kept -800 for brake range, and put 3000 for acceleration range.

I made remote calibration, and the test was a success. Range on remote is much better (hard to say to what extend, because as we control torque and not speed, it’s still hard to keep a certain constant speed, but this extended range makes it much easier to control the skateboard.

Thanks again !
The2Jakes

To celebrate this I’ve designed and installed the brake light 😉

[image]

zulkep, 2021-05-08:
tried the spd mode, the scale of the regulators is more linear, with the settings
I_MOT_MAX 13
I_DC_MAX 15
tried TRQ mode with settings
I_MOT_MAX 18
I_DC_MAX 25
the knob scale shrank to 20%, full speed set before switching on the weakening, then a dip of 60%, 20% turning on the weakening of the field

zulkep, 2021-06-13:
I still have the same problem, how to stretch the scale of the gas handle, can the filter be what it is? or add a filter? if I limit the revs, then the throttle stick only works at 5%, the other 95% is a useless move
[image]

kris86k, 2021-08-22:
I'ts any new idea to make start more smooter in TRQ mode?

zulkep, 2021-08-23:
smoother?

kris86k, 2021-08-23:
I mean very high acceleration in first second. Its like kick.

zulkep, 2021-08-23:
reduce SPEED_COEFFICIENT

kris86k, 2021-08-23:
Can I add this parameter to ADC or change it in DEFAULT_SPEED_COEFFICIENT ?