EFeru/hoverboard-firmware-hack-FOC #168: [feature idea] return absolute motor rotations in UART

beustens, 2021-04-28:
As the motors have hall sensors, it should be relatively easy to track their accumulated rotation angle or ticks, right? This would be very handy for robot odometry.

vamfun, 2021-04-28:
Yes...4 mechanical deg/tick or diameter/90  distance /tick

http://vamfun.wordpress.com
Robot musing blog
iPhone

beustens, 2021-04-28:
Sorry, I was not clear enough about this. I know how to calculate the rotation from ticks, but currently, there is no API/interface to request the ticks from the board, as far as I have seen.

Candas1, 2021-04-28:
Vamfun has implemented position control in his fork.
It has a function to calculate motor position.

Vamfun, are you counting after 360 degrees ?

busbuyukefe, 2021-04-28:
I was insipred by [hoverboard-driver by @alex-makarov](https://github.com/alex-makarov/hoverboard-driver) and tried doing something similar the FOC firmware. I didn't had the time to test it but diff-drive-controller might be OK just with the wheel speed as input.
And if you want to use diff-drive-controller your firmware needs to get data as RPM values for each wheel instead of speed and steer coefficents, to do that you just need to change [this line](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/36953c6fe9ba90593d21150113459b0a4bb4e312/Src/main.c#L293) with:
`cmdL = steer;    // ADC1
cmdR = speed;  // ADC2`
(thanks @Candas1 !)

I think you can use [this simple C program](https://gist.github.com/busbuyukefe/683de596f6a954a2110724c14e5636e1) that I wrote to decode the serial feedback data received from the hoverboard because I think it might be helpful if you want to write a ROS module. But you need to power-up the hoverboard mainboard after you run the program because the program has can't offset if the start frame isn't correct.

juodumas, 2021-04-28:
@beustens it's already reported via serial as 4th and 5th values (speedR, speedL). Here is an example using Python (use `VARIANT_USART` when compiling):

```python
from typing import Tuple
from time import time
import asyncio
import logging
import math
import struct

class RobotInfo():
    def __init__(self, wheel_diameter_cm: float) -> None:
        self.wheel_length_km: float = (wheel_diameter_cm / 100000) * math.pi

        #
        # Stats obtained from hoverboard controller serial
        #
        self._rpml: int = 0
        self._rpmr: int = 0
        self._battery_voltage: float = 0
        self._board_temp: float = 0

    def get_speed_steer(self) -> Tuple[int, int]:
        # TODO get speed & steer values from somewhere
        return 0, 0

    def set_serial_stats(self, rpml: int, rpmr: int, battery_voltage: float, board_temp: float) -> None:
        # Change rotation sign.
        self._rpml = rpml * -1
        self._rpmr = rpmr * -1

        self._battery_voltage = battery_voltage
        self._board_temp = board_temp

    def get_measured_speed(self) -> Tuple[float, float]:
        return (
            round(self.wheel_length_km * self._rpml * 60, 2),
            round(self.wheel_length_km * self._rpmr * 60, 2),
        )

class SerialProtocol(asyncio.Protocol):
    def __init__(self, robot: RobotInfo) -> None:
        super().__init__()
        self.robot = robot
        self.log: logging.Logger = logging.getLogger('robo.serial')
        self.key: int = 0xABCD
        self.now: int = round(time() * 1000)
        self.last_sent_at: int = self.now
        self.send_interval: int = 100  # ms

    def connection_made(self, transport) -> None:
        self.transport = transport
        self.log.debug('serial open')
        transport.serial.rts = False

    def data_received(self, data: bytes) -> None:
        self.now = round(time() * 1000)

        start: int = 0
        cmd1: int = 0
        cmd2: int = 0
        speedr: int = 0
        speedl: int = 0
        batv: int = 0
        boardt: int = 0
        cmdled: int = 0
        checksum: int = 0

        try:
            start, cmd1, cmd2, speedr, speedl, batv, boardt, cmdled, checksum = struct.unpack('HhhhhhhHH', data)
        except struct.error as e:
            self.log.warning('skipped invalid serial data: %s', e)

        checksum_computed = start ^ cmd1 ^ cmd2 ^ speedr ^ speedl ^ batv ^ boardt ^ cmdled
        if checksum_computed < 0:
            checksum_computed += 2**16

        if start == self.key and checksum == checksum_computed:
            self.robot.set_serial_stats(speedl, speedr, batv / 100, boardt / 10)
        else:
            self.log.warning('skipped invalid serial data: checksum error')

        if self.now - self.last_sent_at >= self.send_interval:
            self.last_sent_at = self.now
            self.send_command()

    def connection_lost(self, exc) -> None:
        self.log.debug('serial port closed')
        asyncio.get_event_loop().stop()

    def send_command(self) -> None:
        speed: int
        steer: int
        checksum: int

        speed, steer = self.robot.get_speed_steer()
        checksum = self.key ^ steer ^ speed
        if checksum < 0:
            checksum += 2**16
        # self.log.debug("Sending speed=% 4d steer=% 4d", speed, steer)
        out_data: bytes = struct.pack('HhhH', self.key, steer, speed, checksum)
        self.transport.write(out_data)

if __name__ == "__main__":
    import asyncio
    import serial.aio  # type: ignore

    wheel_diameter_cm = 10 * 2.54
    serial_port = '/dev/ttyUSB0'
    serial_baudrate = 115200

    robot = RobotInfo(wheel_diameter_cm)
    loop = asyncio.get_event_loop()
    serial_async = serial.aio.create_serial_connection(
        loop, lambda: SerialProtocol(robot), serial_port, serial_baudrate
    )
    loop.run_until_complete(serial_async)
    loop.run_forever()
    loop.close()
```

beustens, 2021-04-28:
I know how to read out the values. But these are just the current speeds, I need the **accumulated positions** of each wheel.

vamfun, 2021-04-28:
Yes...My PID servo fork has cumulative position.  https://github.com/vamfun/Hoverboard-Hack-FOC-PID-Servo
The code is in control.c

void Motor_Pos(){
static int motAngleLeftLast = 0;
static int motAngleRightLast = 0;	
static int cycleDegsL = 0 ;  //wheel encoder roll over count in deg
static int cycleDegsR = 0 ;  //wheel encoder roll over count in deg	
int diffL = 0;
int diffR = 0;
static int cnt = 0;	
	if(cnt == 0) 
	{
		motAngleLeftLast = motAngleLeft ;
		motAngleRightLast = motAngleRight ;
	}
	diffL = motAngleLeft - motAngleLeftLast ;
	//if(cnt % 25 == 0 ) {printf( " diffL :%i, diffR :%i , motAngleRightLast :%i ",diffL , diffR, motAngleRightLast );}
	
if( diffL < -180 ) { cycleDegsL = cycleDegsL + 360; }
else 
if(diffL > 180) { cycleDegsL = cycleDegsL  - 360;}
MotorPosL = motAngleLeft + cycleDegsL;

	diffR = motAngleRight - motAngleRightLast ;
if( diffR < -180 ) { cycleDegsR = cycleDegsR + 360; }
else 
if(diffR > 180) { cycleDegsR = cycleDegsR  - 360;}
MotorPosR = motAngleRight + cycleDegsR;
motAngleLeftLast = motAngleLeft;
motAngleRightLast = motAngleRight;
cnt++;

}
http://vamfun.wordpress.com
Robot musing blog
iPhone

Candas1, 2022-01-02:
Please check this [project](https://github.com/alex-makarov/hoverboard-driver)

closed 2022-01-02 by Candas1