EFeru/hoverboard-firmware-hack-FOC #169: Red mainboard - cannot connect via ST-Link V2

swhof123, 2021-04-28:
Hi all, 
many thanks for all the great work done here! 

I was planning to use the hoverboard as strong base for an autonomous law mower (combining it with the work done by [Ardumover](https://www.ardumower.de/de/))
However, I already get stuck at the first step: Connecting the hoverboard with my mac. 

What I have done so far:
- Solder for pins to the SWD programming holes. I check the Voltage level and get 3.3V between Pin 1 & 3 (counted from the top) and 2.6V between 2 & 3 and 4 & 3
- Connected SWCLK, GND, SWDIO to ST-Link V2. Did not connect 3.3.V
- Downloaded and started STM32 Cube Programmer (thanks to [this post](https://www.carminenoviello.com/2015/09/09/running-stm32cubemx-macos-finally/), I understood that I had to run the exe file as jar file)
- Updated the firmware of ST-Link V2
- Connected the battery.
- Powered on the hoverboard
- Clicked on "connect" and got "No STM32 device found". I tried several combinations of frequency, mode, reset mode, but the result remained the same.  

Also, I tried: 
- Disconnected the battery.
- Connected SWCLK, GND, SWDIO and 3.3.V to ST-Link V2
- Powered on the hoverboard
- Clicked on "connect" and also got an error. I tried several combinations of frequency, mode, reset mode, but the result remained the same.  

I check the board, it has an STM32F103 chip on it. 

Any help would be greatly appreciated.

[image]
[image]

Candas1, 2021-04-29:
Hi,

When you power the board on is it beeping ?
You need to make sure you soldered the headers well.

swhof123, 2021-04-29:
Thanks Candas 1 for your feedback.

Board beeps "tu-tü-dü" when I power it on. Same sound it made before I started tinkering with it.
LED on main board is flashing slowly.

I am not so sure about my soldering, below a pic of how is looks. But as I can measure voltage, I supposed it is ok. 

I just re-measured agains battery GND and found something surprising:
- Pin 1: 3.3 V as expected
- Pin 2: 0V
- Pin 3: 2.6 V, I expected 0V for GND
- Pin 4: 0V
(In my earlier measurement, I had misinterpreted the voltage direction between Pin 2 & 3 and 4 & 3).
What do these measurements mean?
Are the pin positions swapped on this board? 

[image]

EmanuelFeru, 2021-04-29:
It can be that the pin mappings are swapped. I suggest checking with the datasheet and a multimeter set on continuity to confirm and check the mapping.

swhof123, 2021-04-29:
From looking at the datasheet and the pics from other boards, I could not see any difference in the pin mapping. 
I did some continuity testing and indeed, there was one pin without continuity. 
Afterwards, however, there was a little discharge noise and now the board is not working anymore. 
So I guess this issue can be considered resolved. 
Can you recommend me a board that is known to be working with this firmware and that can be purchases in Germany?

EmanuelFeru, 2021-04-29:
So, you managed to connect to is and flash?
I think locally is probably cheaper to buy an used Hoverboard and use the parts of it or if you can wait a few weeks then Aliexpress.

swhof123, 2021-04-30:
Nope, unfortunately I killed the board before having any success. 
I'll try again with a new board. 
Thanks anyway for your support!

closed 2021-04-30 by swhof123

mrlunk, 2021-05-12:
Ground yourself to the mainboard ground wire on the battery connector when working on it !!!!
The mainboard contains several powerfull condensators that can hold charge even when the battery is disconnected !

The shock won't kill you but it will fry your controller board.

After disconnecting the Battery, hold the powerbutton for 2 seconds and you will hear it trying to start up on the leftover charge from the Condensators.
If you remove the power switch cable and short it with a breadboard wire or jumper this waymost of the condensators will be drained all the time.

I work with a anti-static wristband and a 5mm real rubber mat under my desk and chair aswel...
image: `https://twitter.com/MrLunk/status/1392396591936516097`

After disconnecting the Battery, hold the powerbutton for 2 seconds and you will hear it trying to start up on the leftover charge from the Condensators.
If you remove the power switch cable and short it with a breadboard wire or jumper this waymost of the condensators will be drained all the time.

PL.

swhof123, 2021-05-13:
Thanks for the answer, Peter. 
It was probably the charge left in the condensators/capacitors that caused the mainboard to fry. 
Will be more careful from now on.

Anyhow, got myself a new used board, found a green mainboard and the flashing worked like a charm. 
Now, my board is controlled via WIFI, using an ESP32 and the UART Variant. 

I have one or two follow-up questions. 
Where is the best place to ask those? I do not want to spam the git issues with my noob questions.

vamfun, 2021-05-14:
Before giving up on you board might try doing a memory erase per #129 issue.   Saved me from throwing away a board when it would not connect with STM-LINK.

farev, 2023-09-12:
I have had the same issue when trying to connect my board via the st-link, I have also killed two boards trying different solution. The solution was that I only needed to update the software on the st-link itslef. Make sure that the st-link is connected just to your computer, if it is connected to the motherboard with even just one cable it won't work, on the STM32 app you should see an option that says "firmware upgrade" click on it, it should open a new window, choose the connected device and click upgrade. After doing this the st-link connected without a problem to my board.

Candas1, 2023-09-12:
You cannot kill those boards like that.

farev, 2023-09-12:
I killed the boards because of mistakes I made when trying to connect it to the st-link. I think it had to do with the charge that stays on the board even after it is disconnected.

Candas1, 2023-09-12:
Ah ok, after disconnecting the power supply,  press the ON/OFF button to empty the charge before manipulating the board.
While flashing you cannot brick the boards