EFeru/hoverboard-firmware-hack-FOC #171: Q: Is there an easy way to reverse throttle (PWM) ?

mrlunk, 2021-05-06:
Hi there :)

1st off, thanks and much appreciation for your work !!

I am using PWM-control version in Speed mode...

Is there a way to reverse the throttle input so that forward becomes backwards and vice verca ?

In my build I can't swap the motors easily anymore but I realise that it handles better when driving the other way.

Greetings,
Peter Lunk

Candas1, 2021-05-06:
Hi,

You could probably even do that from your transmitter without touching the firmware.

You can also uncomment those 2 lines
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/0709e4cb62ef1d0bfedfc6bc3187614d5862acd6/Inc/config.h#L438

mrlunk, 2021-05-06:
Nevermind found it ;)

`  // #define INVERT_R_DIRECTION`
`  // #define INVERT_L_DIRECTION`

PL.

mrlunk, 2021-05-06:
THanks for your reply Candas1 ;)

mrlunk, 2021-05-06:
Hmm changing that in the Firmware config code works for driving direction...
But the steering stays oriented as if it was still driving the old way around,
resulting in inverted steering now grmpf...
Any ideas welcome.

Sincerely,
Peter Lunk

Candas1, 2021-05-06:
You can also use a negative speed and steer coefficient.
Check nunchuk variant for example

mrlunk, 2021-05-06:
investigating ;) ty

Candas1, 2021-05-07:
No, try 49152 value

mrlunk, 2021-05-07:
Oof ?

I do not understand that..
Is there a place where I can find more on this this logic ?
I'd like to learn :) 

PL.

mrlunk, 2021-05-07:
In code it says:
`[0, 65535] = [-2.0 - 2.0]`
Should that perhaps then be:
`[0, 65535] = [2.0 to -2.0]`

so not: 
-2 to 2 : 0 to 65535
But: 
2 to -2 : 0 to 65535

Then I understand better ;)

PL.

mrlunk, 2021-05-07:
omfg...
I am making some serious thinking errors :P
I understand now .
Thanks for your help @Candas1 ! 👍 
PL.