EFeru/hoverboard-firmware-hack-FOC #172: Question\problem with second wheel

Vittaly, 2021-05-08:
Good day! I had spend all day and have fail with my kickbike.
 I tried to configure car version abd what I have:
Left wheel work fine (not rotate on start and starting after potenciometer moved)
Right wheel start rotate after power on and not stop. If I stop it, it start rotate in revers side. In debug out I see comR and comL = 0 

Please help me.

EmanuelFeru, 2021-05-08:
It can be that you have to swap motor phases for that motor since there is a mismatch in the wire colors.

Vittaly, 2021-05-08:
Thank you. But I check it thee time. And I think  the mistake on phases get only dire tion problem, but not incorect start process. Or not?

vamfun, 2021-05-08:
As @Сandas recommended...make sure you try all combinations of motor phases... it’s easy to do and might fix problem.  The wrong phases can exhibit many more effects than just changing sign in my experience.    

http://vamfun.wordpress.com
Robot musing blog
iPhone

Vittaly, 2021-05-09:
Good night! I had tried again and it works now. The phase wares conected as its shound be (by colors). The one change that I done is moving pull down resistors to a pcb from a ware. I not understed why it works now, but I happy. Thenk you every body.

closed 2021-05-09 by Vittaly